espressif/esp-thread-br #84: Issue with ESP-Thread-BR Example on ESP32-S3 WROOM1

vishnuprasadvp12345, 2024-07-23:
Hi all, this is my first time working with the Thread network protocol. I downloaded the ESP-Thread-BR example and flashed my ESP32-S3 WROOM1 micro-controller with the basic Thread board router example code. However, the code keeps resetting, and I am not sure what went wrong. Can anyone help me understand why the code is resetting and how to solve this issue?
[THREADISSUE.txt](https://github.com/user-attachments/files/16346004/THREADISSUE.txt)

zwx1995esp, 2024-07-23:
Hi, @vishnuprasadvp12345 , our example is used for [ESP Thread Border Router Board](https://github.com/espressif/esp-thread-br?tab=readme-ov-file#esp-thread-border-router-board), if you want to run it on `ESP32-S3 WROOM1` board, you need to connect it with an RCP manually(please refer to [here](https://github.com/espressif/esp-thread-br?tab=readme-ov-file#standalone-modules)).

If a uart RCP is used, please connect S3 with RCP via :
```
    S3                  RCP
GPIO17 (UART RX)  <->TX
GPIO18 (UART TX)  <-> RX
GPIO7             <-> RST
GPIO8             <-> GPIO9 (BOOT)
```
Notice: pin 17 and pin 18 have a different driver current with other GPIOs(refer to [ESP32-S3 TRM chapter 6.12, page 489](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf)). If you connect the RCP with S3 using jumpers, please change the S3 Tx/Rx gpios(for example, pin 4 and pin 5).

Also the default console configuration for S3 is `ESP_CONSOLE_USB_SERIAL_JTAG`, so make sure the `USB` port(not `UART` port) is connected to the host.

vishnuprasadvp12345, 2024-07-24:
I am using the ESP32-C6 for Thread networking. I flashed the RCP example code to the ESP-IDF and the Border Router example code to the ESP32-S3. I also connected the UART pins, but the code keeps resetting.

zwx1995esp, 2024-07-24:
Hi, @vishnuprasadvp12345 Ok, could you please share these informations which are useful for debugging?

1. How did you connect the C6 with S3? Could you please share the pin map you used?
2.Could you please share the sdkconfig you used for S3?
3.Which port did you connect on to host, the `UART` or `USB` port on S3?

vishnuprasadvp12345, 2024-07-24:
[report.txt](https://github.com/user-attachments/files/16357739/report.txt)

zwx1995esp, 2024-07-24:
Hi @vishnuprasadvp12345 As this mentioned:

> Hi, @vishnuprasadvp12345 , our example is used for [ESP Thread Border Router Board](https://github.com/espressif/esp-thread-br?tab=readme-ov-file#esp-thread-border-router-board), if you want to run [...]

1. Our default config is `ESP_CONSOLE_USB_SERIAL_JTAG` enabled, so please connect the S3 `USB` port to host.

2.  pin 17 and pin 18 have a different driver current with other GPIOs for S3, please try other gpios like pin 4 and pin 5.

Please have a try.

vishnuprasadvp12345, 2024-07-24:
I'm used to flashing the ESP-IDF examples, specifically the OpenThread ot_br example code, on the ESP32-C6. I can send all the commands to the board router, and it's working on the ESP32-S3. However, I'm encountering the following warnings and errors

W(2596) OPENTHREAD: [W] P-RadioSpinel-: Wait for response timeout
W(4606) OPENTHREAD: [W] P-RadioSpinel-: Wait for response timeout
E(4606) OPENTHREAD: [C] P-RadioSpinel-: Failed to reset RCP!
E(4606) OPENTHREAD: [C] Platform------: ResetRcp() at radio_spinel.cpp:233: Failure

zwx1995esp, 2024-07-24:
> I'm used to flashing the ESP-IDF examples, specifically the OpenThread ot_br example code, on the ESP32-C6. I can send all the commands to the board router, and it's working on the ESP32-S3. [...]

Hi, @vishnuprasadvp12345 for the IDF example `ot_br`, we set the uart pin4 and pin5 as default. Please refer to: [How to use example](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br#how-to-use-example) and the [source code](https://github.com/espressif/esp-idf/blob/master/examples/openthread/ot_br/main/esp_ot_config.h#L40)

So could you please check the pin map you used?

vishnuprasadvp12345, 2024-07-24:
[uartconfig.txt](https://github.com/user-attachments/files/16361343/uartconfig.txt)
Are there any other pins I need to connect?

zwx1995esp, 2024-07-24:
> [uartconfig.txt](https://github.com/user-attachments/files/16361343/uartconfig.txt) Are there any other pins I need to connect?

As you shown in the config, so, the map between S3 and C6 should be like:
```
    S3                  C6
GPIO4 (UART RX)  <-> GPIO16(UART TX)
GPIO5 (UART TX)  <-> GPIO17(UART RX)
3V3              <-> 3V3
GND              <-> GND
```
Did you connect the S3 with C6 like this?

vishnuprasadvp12345, 2024-07-25:
I'm connecting the same way, but I am powering it through the USB port. In this case, I encountered a reset issue in the S3 OT_BR example code:
I (560) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (560) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (580) OPENTHREAD: spinel UART interface initialization completed
I (560) main_task: Returned from app_main()
W (2600) OPENTHREAD: [W] P-RadioSpinel-: Wait for response timeout
W (4610) OPENTHREAD: [W] P-RadioSpinel-: Wait for response timeout
E (4610) OPENTHREAD: [C] P-RadioSpinel-: Failed to reset RCP!
E (4610) OPENTHREAD: [C] Platform------: ResetRcp() at radio_spinel.cpp:233: Failure
[image]

zwx1995esp, 2024-07-25:
Oh, please use this config on the RCP side(C6) then have a try:
```
   #define ESP_OPENTHREAD_DEFAULT_HOST_CONFIG()                    \
      {                                                           \
   ...(skip the same config)...
                    },                                              \
                .rx_pin = UART_PIN_NO_CHANGE,               \
                .tx_pin = UART_PIN_NO_CHANGE,               \
            },                                                      \
    }

```

BTW: for the example `OT_BR` in idf branch, the default console config is `ESP_CONSOLE_UART_DEFAULT`, so please connect the `UART` port onto host.

vishnuprasadvp12345, 2024-07-26:
<0x1b>[0;33mW(11663) OPENTHREAD:[W] P-RadioSpinel-: Wait for response timeout<0x1b>[0m
<0x1b>[0;32mI(11663) OPENTHREAD:[I] P-RadioSpinel-: Hardware reset RCP successfully<0x1b>[0m
<0x1b>[0;33mW(13683) OPENTHREAD:[W] P-RadioSpinel-: Wait for response timeout<0x1b>[0m
<0x1b>[0;33mW(13683) OPENTHREAD:[W] P-RadioSpinel-: RCP failure detected<0x1b>[0m
<0x1b>[0;33mW(13683) OPENTHREAD:[W] P-RadioSpinel-: Trying to recover (1/3)<0x1b>[0m
<0x1b>[0;33mW(15703) OPENTHREAD:[W] P-RadioSpinel-: Wait for response timeout<0x1b>[0m
<0x1b>[0;32mI(15703) OPENTHREAD:[I] P-RadioSpinel-: Hardware reset RCP successfully<0x1b>[0m
The RCP is not responding. I checked, and the UART pins are the same.

zwx1995esp, 2024-07-29:
Hi, let's clear all code and your config and try again:
first let's compile a RCP using ESP32-C6 using the tag v5.2.2.
Step 1: Prepare the idf branch:
```
$ cd <your-idf-dir>
$ git checkout v5.2.2
$ git submodule update --init --recursive
$ ./install.sh
$ . ./export.sh
$ cd examples/openthread/ot_rcp
```
Step 2: remove all pre-built files and configures:
```
$ rm -rf build sdkconfig 
```
Step 3: check no changes for the idf:
```
$ git status
(if you have any changes, please revert them)
```
Step 4: set target and build
```
idf.py set-target esp32c6
idf.py build
idf.py -p <your-local-port-for-c6> flash 
```

Then let's build and run BR on S3
Step 1: Goto BR example directory, clean the build files and configs:
```
cd <your-idf-dir>/examples/openthread/ot_br
$ rm -rf build sdkconfig dependencies.lock
```
Step 2: check no changes for the idf:
```
$ git status
(if you have any changes, please revert them)
```
Step 3: set target and build
```
idf.py set-target esp32s3
idf.py build
```

Then connect the S3 with C6(Notice: please first flash C6 board with RCP firmware then connect it using the jumpter with S3)
```
    S3                  C6
GPIO4 (UART RX)  <-> TX(UART TX)
GPIO5 (UART TX)  <-> RX(UART RX)
3V3              <-> 3V3
GND              <-> GND
```
like this picture:

[image]
Then flash BR example for S3
```
$ idf.py -p <your-local-port-for-s3> flash monitor
```
(No need for changing any codes to run the BR example on S3)
Then you will get the BR running on S3:
```
I (544) main_task: Calling app_main()
I (564) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (574) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (584) OPENTHREAD: spinel UART interface initialization completed
I (574) main_task: Returned from app_main()
I(594) OPENTHREAD:[I] P-SpinelDrive-: co-processor reset: RESET_POWER_ON
E(594) OPENTHREAD:[C] P-SpinelDrive-: Software reset co-processor successfully
I(654) OPENTHREAD:[I] ChildSupervsn-: Timeout: 0 -> 190
I(674) OPENTHREAD:[I] Settings------: Read NetworkInfo {rloc:0x3400, extaddr:1af7fa332fc3cc29, role:leader, mode:0x0f, version:5, keyseq:0x0, ...
I(684) OPENTHREAD:[I] Settings------: ... pid:0x660d450a, mlecntr:0x2f1d, maccntr:0x1b5c, mliid:4d563e9bf965bd58}
I (694) OPENTHREAD: OpenThread attached to netif
> > help
ba
bbr
br
bufferinfo
ccathreshold
ccm
channel
child
...
udpsockserver
wifi
Done
>
```

vishnuprasadvp12345, 2024-07-31:
its working thank you so much @zwx1995esp

closed 2024-07-31 by chshu