espressif/esp-thread-br #86: NAT64 Translator not starting

goossensbas, 2024-08-07:
Hello,
I installed the OTBR firmware on the espressif border router dev board.

When i enable NAT64 on the border router only the prefixmanager is visible when i check it with 'nat64 state'. 
The translator is not there.
I checked the menuconfig settings, but everything seems OK there.

How can i enable the translator?

Thanks!

zwx1995esp, 2024-08-07:
For the NAT64 translator feature, we enable it by default. No needs for doing any other things. You can have a try with this codelab: [3.4. NAT64](https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/nat64.html#nat64) in our docs.

goossensbas, 2024-08-14:
@zwx1995esp Thank you for your reply.

The codelab is already using DNS over IPV4, but i cannot use IPV4 on my border router.
When i start a new network, and i check with "nat64 state", the translator is missing. I only have the PrefixManager.

Is there something i'm missing? Is version 1.0 from openthread.io the correct one

These are the steps i did to install the firmware (from [openthread.io](https://openthread.io/guides/border-router/espressif-esp32)):
(I'm using esp-idf v5.1.2)

```
$ git clone -b v1.0 --recursive https://github.com/espressif/esp-thread-br.git
$ cd esp-thread-br/examples/basic_thread_border_router
$ idf.py menuconfig
```

In the menuconfig i enabled the following options. The other options i did not change:
- ESP thread Border Router Example / Enable automatic start
- ESP thread Border Router Example / Enable web server
- Example Connection Configuration / SSID
- Example Connection Configuration / password
`
$ idf.py build flash monitor /dev/ttyACM1`

After starting a new network i try to ping 8.8.8.8, but it does not work.

zwx1995esp, 2024-08-15:
NAT64 is using for thread device which do not has wifi or eth link to access the cloud or other local ipv4 address. BR will forward the packets between the IPv4 address host and the thread end device. For the BR, if you want to access the IPv4 address, no need to use NAT64(because BR has wifi or eth link).

Also you can not `ping` a ipv4 address using a Thread layer ping command. Please refer to our [BR docs FAQ](https://docs.espressif.com/projects/esp-thread-br/en/latest/qa.html#q1-why-can-t-i-access-the-host-from-the-br-device-using-the-ping-command) to get more informations.

chshu, 2024-09-06:
@goossensbas do you have any follow up questions? or could you close the issue?

closed 2024-09-19 by chshu