espressif/esp-thread-br #87: Using OpenThread CoAP API on ESP BR Router

cedriczou13, 2024-08-07:
### Question: Using OpenThread CoAP API on ESP BR Router

Hi,

I am currently working with the ESP BR router and attempting to utilize the OpenThread CoAP API to send a CoAP message to the cloud. However, I am encountering some errors in the process.

**Details:**
- **Device:** ESP BR Router
- **API:** OpenThread CoAP API
- **Objective:** Send a CoAP message to a cloud service

**Issue:**
When I try to send a CoAP message using the OpenThread CoAP API, I am receiving an error. Here are the specific details of the error message and the relevant portion of my code:

```c
void coap_send_mess(void)
{
    otError error = OT_ERROR_NONE;
    otMessage *p_request;
    otMessageInfo message_info;

    otInstance *p_instance = esp_openthread_get_instance();

    otIp6Address synt_ipv6_addr;
    otIp4Address tmp_ipv4_addr;
    const char *aAddress = "167, 99, 85, 188";

    memset(&message_info, 0, sizeof(message_info));

    otIp4AddressFromString(aAddress, &tmp_ipv4_addr);

    if (OT_ERROR_NONE != otNat64SynthesizeIp6Address(esp_openthread_get_instance(), &tmp_ipv4_addr, &message_info.mPeerAddr))
    {
        printf("Error synthesizing IPv6 address from IPv4 address\n");
    }

    char buf[OT_IP6_ADDRESS_STRING_SIZE];
    otIp6AddressToString(&message_info.mPeerAddr, buf, OT_IP6_ADDRESS_STRING_SIZE);
    printf("Address [%s]\n", buf);

    do
    {
        p_request = otCoapNewMessage(p_instance, NULL);
        if (p_request == NULL)
        {
            printf("Failed to allocate message for CoAP Request\r\n");
            break;
        }

        otCoapMessageInit(p_request, OT_COAP_TYPE_NON_CONFIRMABLE, OT_COAP_CODE_POST);

        error = otCoapMessageAppendUriPathOptions(p_request, "brhc");
        if (error != OT_ERROR_NONE)
        {
            printf(" error otCoapMessageSetPayloadMarker\r\n");
        }

        error = otCoapMessageSetPayloadMarker(p_request);
        if (error != OT_ERROR_NONE)
        {
            printf(" error otCoapMessageSetPayloadMarker\r\n");
        }
        uint8_t command = 5;
        error = otMessageAppend(p_request, &command, sizeof(command));
        if (error != OT_ERROR_NONE)
        {
            break;
        }

        message_info.mPeerPort = OT_DEFAULT_COAP_PORT;

        error = otCoapSendRequest(p_instance, p_request, &message_info, NULL, NULL);
    } while (false);

    if (error != OT_ERROR_NONE && p_request != NULL)
    {
        printf("Failed to send CoAP Request: %d\r\n", error);
        otMessageFree(p_request);
    }
    else
    {
        printf(" sent CoAP Request: \r\n");
    }
}
```

**Error Message:**
```
E (53629) OPENTHREAD: Failed to Send UDP message, err: -12
 sent CoAP Request:
```

I would appreciate any guidance or insights on what might be causing this issue and how I can successfully send a CoAP message to the cloud using the OpenThread CoAP API on the ESP BR router.

Thank you in advance for your help!

Best regards,
Cedric

zwx1995esp, 2024-08-08:
Hi, @cedriczou13 Your issue is more like this question in our BR docs: https://docs.espressif.com/projects/esp-thread-br/en/latest/qa.html#q1-why-can-t-i-access-the-host-from-the-br-device-using-the-ping-command.

You can call the ot APIs to create a COAP message and add the detailed information you like in the message. But for sending this message to the cloud on BR device, using any openthread APIs is not a reasonable choose. There is a WiFi(or Ethernet) interface on the BR, so the better choose is using the lwip UDP API to send the COAP message. 

BTW, your code might work for the other thread end devices to access the cloud via a NAT64-enabled thread BR. I mean in your thread network, this code might work on the device which only has the thread netif. The message will be forward to cloud via BR. Only a risk here: if the function `coap_send_mess` is not processed in the ot task, you need to acquire the ot lock before calling OT APIs. Please refer to here: https://github.com/espressif/esp-idf/blob/master/components/openthread/include/esp_openthread_lock.h#L48.

And I have a question: Does the BR device have to access the cloud through COAP request? Or your project just needs the BR forward these requests from Thread ED to cloud?

chshu, 2024-09-06:
@cedriczou13 Do you have any update on this issue?

closed 2024-09-19 by chshu