espressif/esp-thread-br #88: RCP update can't work

GASEN1216, 2024-08-21:
We make the board with ESP32-S3 and ESP32-H2.

And H2 pin as follow:
[image]

IDF: v5.2.2

I run the examples basic_thread_border_router, and try to update the RCP. But it prompts the following error:

```I (26) boot: ESP-IDF v5.2.2-dirty 2nd stage bootloader
I (27) boot: compile time Aug 21 2024 15:33:02
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (80) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (87) boot:  3 ota_0            OTA app          00 10 00020000 00190000
I (94) boot:  4 ota_1            OTA app          00 11 001b0000 00190000
I (102) boot:  5 web_storage      Unknown data     01 82 00340000 00019000
I (110) boot:  6 rcp_fw           Unknown data     01 82 00359000 000a0000
I (117) boot: End of partition table
I (121) esp_image: segment 0: paddr=00020020 vaddr=3c110020 size=56de4h (355812) map
I (194) esp_image: segment 1: paddr=00076e0c vaddr=3fc98600 size=05440h ( 21568) load
I (199) esp_image: segment 2: paddr=0007c254 vaddr=40374000 size=03dc4h ( 15812) load
I (203) esp_image: segment 3: paddr=00080020 vaddr=42000020 size=10bdd8h (1097176) map
I (405) esp_image: segment 4: paddr=0018be00 vaddr=40377dc4 size=107e8h ( 67560) load
I (429) boot: Loaded app from partition at offset 0x20000
I (429) boot: Disabling RNG early entropy source...
I (430) cpu_start: Multicore app
I (442) cpu_start: Pro cpu start user code
I (442) cpu_start: cpu freq: 160000000 Hz
I (443) cpu_start: Application information:
I (443) cpu_start: Project name:     esp_ot_br
I (443) cpu_start: App version:      v5.2.2-dirty
I (443) cpu_start: Compile time:     Aug 21 2024 15:43:22
I (443) cpu_start: ELF file SHA256:  8f41d4baf...
I (443) cpu_start: ESP-IDF:          v5.2.2-dirty
I (444) cpu_start: Min chip rev:     v0.0
I (444) cpu_start: Max chip rev:     v0.99
I (444) cpu_start: Chip rev:         v0.1
I (444) heap_init: Initializing. RAM available for dynamic allocation:
I (445) heap_init: At 3FCB1710 len 00038000 (224 KiB): RAM
I (445) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (445) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (445) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (447) spi_flash: detected chip: generic
I (447) spi_flash: flash io: dio
I (447) sleep: Configure to isolate all GPIO pins in sleep state
I (448) sleep: Enable automatic switching of GPIO sleep configuration
I (448) main_task: Started on CPU0
I (458) main_task: Calling app_main()
I (538) example_connect: Start example_connect.
I (538) pp: pp rom version: e7ae62f
I (548) net80211: net80211 rom version: e7ae62f
I (558) wifi:wifi driver task: 3fcbd258, prio:23, stack:6144, core=0
I (558) wifi:wifi firmware version: 3e0076f
I (558) wifi:wifi certification version: v7.0
I (558) wifi:config NVS flash: enabled
I (558) wifi:config nano formating: enabled
I (558) wifi:Init data frame dynamic rx buffer num: 32
I (558) wifi:Init static rx mgmt buffer num: 5
I (558) wifi:Init management short buffer num: 32
I (558) wifi:Init dynamic tx buffer num: 32
I (558) wifi:Init static tx FG buffer num: 2
[... 84 lines cut ...]
            .tx_pin = CONFIG_PIN_TO_RCP_RX,                \
        },                                                 \
    }
#endif // CONFIG_OPENTHREAD_RADIO_SPINEL_UART OR  CONFIG_OPENTHREAD_RADIO_SPINEL_SPI

#define ESP_OPENTHREAD_RCP_UPDATE_CONFIG()                                                                           \
    {                                                                                                                \
        .rcp_type = RCP_TYPE_ESP32H2_UART, .uart_rx_pin = CONFIG_PIN_TO_RCP_TX, .uart_tx_pin = CONFIG_PIN_TO_RCP_RX, \
        .uart_port = 1, .uart_baudrate = 115200, .reset_pin = CONFIG_PIN_TO_RCP_RESET,                               \
        .boot_pin = CONFIG_PIN_TO_RCP_BOOT, .update_baudrate = 460800, .firmware_dir = "/rcp_fw/ot_rcp",             \
        .target_chip = ESP32S3_CHIP,                                                                                 \
    }
#define ESP_OPENTHREAD_DEFAULT_HOST_CONFIG()                        \
    {                                                               \
        .host_connection_mode = HOST_CONNECTION_MODE_CLI_USB,       \
        .host_usb_config = USB_SERIAL_JTAG_DRIVER_CONFIG_DEFAULT(), \
    }
#endif

#define ESP_OPENTHREAD_DEFAULT_PORT_CONFIG()                                            \
    {                                                                                   \
        .storage_partition_name = "nvs", .netif_queue_size = 10, .task_queue_size = 10, \
    }
```
what is the problems?

gytxxsy, 2024-08-22:
We recommend using the ESP Thread Border Router Board instead of connecting the ESP32-S3 and ESP32-H2 with wires. If you have to do this and encounter the `Failed to connect to RCP` issue, you can try changing the `Pin to RCP TX` and `Pin to RCP RX` pins on the S3, for example:
* Pin to RCP TX: 4
* Pin to RCP RX: 5

GASEN1216, 2024-08-23:
> If you make the board with ESP32-S3 and ESP32-H2, and encounter the `Failed to connect to RCP` issue, you should change the `Pin to RCP TX` and `Pin to RCP RX` pins on the S3, for example:如果您使用 ESP32-S3 和 ESP32-H2 制作开发板，并遇到 `Failed to connect to RCP` 问题，您应该将 S3 上的 `Pin 更改为 RCP TX` 和 `Pin 更改为 RCP RX` 引脚，例如ample：
> 
> * Pin to RCP TX: 4引脚到 RCP TX：4
> * Pin to RCP RX: 5引脚到 RCP RX：5
> 
> Refer to the [esp-thread-br README.md](https://github.com/espressif/esp-thread-br?tab=readme-ov-file#standalone-modules) to get more information.请参考 [esp-thread-br README.md](https://github.com/espressif/esp-thread-br?tab=readme-ov-file#standalone-modules) 获取更多信息。

Configuration in esp-thread-br：
[image]
Configuration in rcp：
[image]
Still showing this error：E (9483) RCP_UPDATE: esp_rcp_update(247): Failed to connect to RCP

zwx1995esp, 2024-08-23:
According to the docs:https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#set-up-the-repositories, we recommend using IDF v5.1.3 with the esp-thread-br tag v1.0. Could you please have a try? If this code base still can not work, could you please take a photo to show how did you connect the S3 with the H2?

GASEN1216, 2024-08-26:
> According to the docs:https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#set-up-the-repositories, we recommend using IDF v5.1.3 with the esp-thread-br tag v1.0. Could you please have a try? If this code base still can not work, could you please take a photo to show how did you connect the S3 with the H2?根据 docs：https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#set-up-the-repositories，我们建议使用 IDF v5.1.3 和 esp-thread-br 标签 v1.0。你能试一试吗？如果这个代码库仍然无法工作，您能否请拍张照片来展示您是如何将 S3 与 H2 连接的？

I tried it and got a new error: 
```
I (15471) RCP_UPDATE: RCP: using update sequence 1
I (15481) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (15481) OPENTHREAD: spinel UART interface initialization completed
I (15481) main_task: Returned from app_main()
W(17491) OPENTHREAD:[W] P-RadioSpinel-: Wait for response timeout
I (17491) gpio: GPIO[7]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (18491) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
W(20501) OPENTHREAD:[W] P-RadioSpinel-: Wait for response timeout
E(20501) OPENTHREAD:[C] P-RadioSpinel-: Failed to reset RCP!
E(20501) OPENTHREAD:[C] Platform------: ResetRcp() at radio_spinel.cpp:233: Failure
```
The board I used: 
[image]

The configuration is the same as above

zwx1995esp, 2024-08-26:
Hi, @GASEN1216 from the picture you shared, you are using the Thread Board Router Board. No more other configurations related to GPIO pins need to be set. Could you please clean all the build files and try again with following the steps in the [docs](https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#build-and-run)? 
Tips: to clean all build files, you can remove the build folder, set target and re-build the BR examples `examples/basic_thread_border_router`.

GASEN1216, 2024-08-26:
> Hi, @GASEN1216 from the picture you shared, you are using the Thread Board Router Board. No more other configurations related to GPIO pins need to be set. Could you please clean all the build files and try again with the following steps in the [docs](https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#build-and-run)? Tips: to clean all build files, you can remove the build folder, set target and re-build the BR examples `examples/basic_thread_border_router`.

Sorry, I know very little about hardware. Thanks for your help. I tried it again with V5.1.3 according to the steps in the docs and it worked.I will close this issue.

GASEN1216, 2024-08-27:
> Hi, @GASEN1216 from the picture you shared, you are using the Thread Board Router Board. No more other configurations related to GPIO pins need to be set. Could you please clean all the build files [...]

Hi, @zwx1995esp 
When I use wifi, RCP update works fine. But when I add ethernet as described in the docs, it stops working:
```
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1758
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbe4
load:0x403cc700,len:0x2d18
entry 0x403c9908
I (26) boot: ESP-IDF v5.1.3-dirty 2nd stage bootloader
I (26) boot: compile time Aug 27 2024 15:50:32
I (26) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (79) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (87) boot:  3 ota_0            OTA app          00 10 00020000 00190000
I (94) boot:  4 ota_1            OTA app          00 11 001b0000 00190000
I (101) boot:  5 web_storage      Unknown data     01 82 00340000 00019000
I (109) boot:  6 rcp_fw           Unknown data     01 82 00359000 000a0000
I (117) boot: End of partition table
I (121) esp_image: segment 0: paddr=00020020 vaddr=3c110020 size=531d4h (340436) map
I (190) esp_image: segment 1: paddr=000731fc vaddr=3fc99100 size=04ecch ( 20172) load
I (195) esp_image: segment 2: paddr=000780d0 vaddr=40374000 size=07f48h ( 32584) load
I (204) esp_image: segment 3: paddr=00080020 vaddr=42000020 size=107a28h (1079848) map
I (399) esp_image: segment 4: paddr=00187a50 vaddr=4037bf48 size=0d170h ( 53616) load
I (420) boot: Loaded app from partition at offset 0x20000
I (420) boot: Disabling RNG early entropy source...
I (421) cpu_start: Multicore app
I (424) cpu_start: Pro cpu up.
I (428) cpu_start: Starting app cpu, entry point is 0x403755fc
0x403755fc: call_start_cpu1 at C:/Users/GASEN/esp/v5.1.3/esp-idf/components/esp_system/port/cpu_start.c:159

I (0) cpu_start: App cpu up.
I (443) cpu_start: Pro cpu start user code
I (443) cpu_start: cpu freq: 160000000 Hz
I (443) cpu_start: Application information:
I (444) cpu_start: Project name:     esp_ot_br
I (444) cpu_start: App version:      v5.1.3-dirty
I (444) cpu_start: Compile time:     Aug 27 2024 15:49:38
I (444) cpu_start: ELF file SHA256:  97e59ab1...
I (444) cpu_start: ESP-IDF:          v5.1.3-dirty
I (445) cpu_start: Min chip rev:     v0.0
I (445) cpu_start: Max chip rev:     v0.99
I (445) cpu_start: Chip rev:         v0.2
I (445) heap_init: Initializing. RAM available for dynamic allocation:
I (446) heap_init: At 3FCB0D18 len 000389F8 (226 KiB): DRAM
I (446) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (446) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (446) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (448) spi_flash: detected chip: gd
I (448) spi_flash: flash io: dio
W (448) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (449) sleep: Configure to isolate all GPIO pins in sleep state
I (449) sleep: Enable automatic switching of GPIO sleep configuration
I (450) app_start: Starting scheduler on CPU0
I (450) app_start: Starting scheduler on CPU1
I (450) main_task: Started on CPU0
I (460) main_task: Calling app_main()
I (530) esp_ot_br: Initializing SNTP
I (550) esp_eth.netif.netif_glue: 02:00:00:12:34:56
[... 22 lines cut ...]
I (6930) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (6930) gpio: GPIO[41]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037bea2
0x4037bea2: esp_cpu_wait_for_intr at C:/Users/GASEN/esp/v5.1.3/esp-idf/components/esp_hw_support/cpu.c:121
```

[image]

The configuration in EXAMPLE_CONNECT_ETHERNET is the same as in the docs
```
Type | W5500 Module 
Stack Size | 2048
SPI Host | SPI2 
SPI SCLK | GPIO21
SPI MOSI | GPIO45 
SPI MISO | GPIO38
SPI CS | GPIO41
SPI Interrupt | GPIO39
SPI SPEED | 36 MHz 
PHY Reset | GPIO40
PHY Address | 1 
```

zwx1995esp, 2024-08-28:
Hi @GASEN1216 It seems like the BR did not receive any bytes from RCP. Could you please double check the configuration related to RCP and BR pin? Did you modify them?

closed 2024-09-01 by GASEN1216