espressif/esp-thread-br #89: How do I auto start thread nnetwork?

Danimal4326, 2024-08-23:
I have CONFIG_OPENTHREAD_BR_AUTO_START enabled and after a reboot, wifi connects automatically, but I still need to run the following commands via cli to start the thread network:
```
ifconfig up
thread start
```

Danimal4326, 2024-08-23:
I did some more debugging and it seems **example_wifi_sta_do_connect** is waiting for a IPV6 address but it never receives one.

zwx1995esp, 2024-08-26:
Hi, could you please share the code base you used? Maybe the commit hash and branch namd of idf and the esp-thread-br repo are very useful to understand what happens. Also did you tried the recommended version related to our [docs](https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#build-and-run)?

Danimal4326, 2024-08-26:
I'm using the HEAD of main for esp-thread-br and tag: v5.3 for esp-idf

chshu, 2024-09-06:
@Danimal4326 If you still have the issue, could you double check the Wi-Fi SSID and Password are correct? Could you share the whole booting logs?

chshu, 2024-09-23:
@Danimal4326 Please reopen if you have any follow up questions.

closed 2024-09-23 by chshu