espressif/esp-thread-br #90: Commissioning fails when using a custom factory and secure_cert partition

hatl, 2024-08-29:
The error occurs when commissioning a device using a custom factory partition and secure_cert partition.

Summary:
- esp-thread-br v1.0 fails most of the times, but not always
- wifi always works
- ZBT1 (SkyConnect) USB dongle always works

For details, see: https://github.com/espressif/esp-matter/issues/1056

hatl, 2024-08-29:
Also fails when using idf v5.3.0 and the main branch of esp-thread-br 6011f07ca9bd31686944b33422d86c1e4d429479
Altough the error is slightly different

[device.log](https://github.com/user-attachments/files/16805527/device.log)
[br.log](https://github.com/user-attachments/files/16805528/br.log)

wqx6, 2024-09-06:
Will the commissioning also fail when not using the custom factory partition and secure_cert partition?

hatl, 2024-09-26:
The commissioning works when using e.g. this one: https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/mfg_binaries/20202020_3841.bin
So it must be related to the custom factory/cert partition

chshu, 2024-09-29:
Closing the issue since it's not related to Thread BR, let's follow up in https://github.com/espressif/esp-matter/issues/1056 for the factory/cert partition.

closed 2024-09-29 by chshu

hatl, 2024-09-29:
> Closing the issue since it's not related to Thread BR, let's follow up in [espressif/esp-matter#1056](https://github.com/espressif/esp-matter/issues/1056) for the factory/cert partition.

how did you come to this conclusion?
since the same procedure works with the ZBT1 stick but does not with the esp-thread-br, I was pretty sure that the issue is caused by the esp-thread-br

chshu, 2024-09-29:
@hatl Let's follow up the issue in https://github.com/espressif/esp-matter/issues/1056