espressif/esp-thread-br #92: Example not building if you checkout w/o v1.0 tag.

JanneKiiskila, 2024-09-09:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Based on the documentation:

https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#set-up-the-repositories

You are asked to checkout specifically the SDK version v5.1.3 but not to check any specific version of the example itself.

SDK:
```
git clone -b v5.1.3 --recursive https://github.com/espressif/esp-idf.git
```

Example:
```
git clone --recursive https://github.com/espressif/esp-thread-br.git
```

However, that picks a the floating tip of main - which as of today at least - gets you a version which will fail the compilation. You must use version v1.0, instead. The [OpenThread instructions](https://openthread.io/guides/border-router/espressif-esp32) have this right:

```
git clone -b v1.0 --recursive https://github.com/espressif/esp-thread-br.git
```

Easy fix - add the `-b v1.0` to the documentation. Document in a README.md or similar with version of the SDK the example assumes.

### Actual behavior (suspected bug)

Build fails.

### Error logs or terminal output

```shell
If you do not use the tagged version the compilation will fail with:

In file included from /home/jannek/openThread/esp32/esp-idf/components/esp_common/include/esp_err.h:13,
                 from /home/jannek/openThread/esp32/esp-idf/components/esp_common/include/esp_check.h:8,
                 from /home/jannek/openThread/esp32/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_rcp_commands.c:11:
/home/jannek/openThread/esp32/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_rcp_commands.c: In function 'esp_openthread_process_rcp_command':
/home/jannek/openThread/esp32/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_rcp_commands.c:30:29: error: void value not ignored as it ought to be
   30 |         ESP_RETURN_ON_FALSE(esp_openthread_rcp_deinit() == ESP_OK, OT_ERROR_FAILED, OT_EXT_CLI_TAG,
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jannek/openThread/esp32/esp-idf/components/esp_common/include/esp_compiler.h:31:25: note: in definition of macro 'unlikely'
   31 | #define unlikely(x)    (x)
      |                         ^
/home/jannek/openThread/esp32/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_rcp_commands.c:30:9: note: in expansion of macro 'ESP_RETURN_ON_FALSE'
   30 |         ESP_RETURN_ON_FALSE(esp_openthread_rcp_deinit() == ESP_OK, OT_ERROR_FAILED, OT_EXT_CLI_TAG,
      |         ^~~~~~~~~~~~~~~~~~~
/home/jannek/openThread/esp32/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_rcp_commands.c:38:29: error: implicit declaration of function 'esp_openthread_rcp_init'; did you mean 'esp_openthread_rcp_deinit'? [-Werror=implicit-function-declaration]
   38 |         ESP_RETURN_ON_FALSE(esp_openthread_rcp_init() == ESP_OK, OT_ERROR_FAILED, OT_EXT_CLI_TAG,
      |                             ^~~~~~~~~~~~~~~~~~~~~~~
/home/jannek/openThread/esp32/esp-idf/components/esp_common/include/esp_compiler.h:31:25: note: in definition of macro 'unlikely'
   31 | #define unlikely(x)    (x)
      |                         ^
/home/jannek/openThread/esp32/esp-thread-br/examples/basic_thread_border_router/managed_components/espressif__esp_ot_cli_extension/src/esp_ot_rcp_commands.c:38:9: note: in expansion of macro 'ESP_RETURN_ON_FALSE'
   38 |         ESP_RETURN_ON_FALSE(esp_openthread_rcp_init() == ESP_OK, OT_ERROR_FAILED, OT_EXT_CLI_TAG,
      |         ^~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[1219/1265] Building C object esp-idf/espressif__mdns/CMakeFiles/__idf_espressif__mdns.dir/mdns.c.objninja: build stopped: subcommand failed.
```
```

### Steps to reproduce the behavior

Follow the steps in: https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#set-up-the-repositories

### Project release version

main

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 22.04

### Shell

Bash

gytxxsy, 2024-09-12:
https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#set-up-the-repositories

The documentation defaults to using the latest `esp-thread-br` commit. If you follow the instructions, the `esp-thread-br` after `git clone` should be on the current latest commit, allowing for successful compilation. If you manually switched to another commit, it may lead to some compilation issues.

You can manually switch to the latest commit and try again. For example, you can switch to the `main` branch. 

Additionally, the esp-thread-br tag: v1.0 has its corresponding esp-idf branch. Please refer to: 
https://github.com/espressif/esp-thread-br/releases/tag/v1.0.

chshu, 2024-09-19:
You can either use the versions specified in the v1.0 release tag:
- IDF v5.1.2
- esp-thread-br commit: https://github.com/espressif/esp-thread-br/commit/b96fbd8e03bf624ab2a84554c17c21fa2592dd6c

Or the versions specified in the latest docs page:
- IDF v5.1.3
- esp-thread-br latest main branch.

Both the above version combinations work without any problem. It also works if you want to build with some different version combination, but need to change some default configurations, for example: https://github.com/espressif/esp-thread-br/issues/94#issuecomment-2359901165.

We are working on a new v1.1 release, which will be based on IDF v5.3.1.

closed 2024-09-19 by chshu