espressif/esp-thread-br #93: Wi-Fi not staying connected - issue wifi connect and device reboots

JanneKiiskila, 2024-09-11:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I would assume the ESP32 can connect to a Wi-Fi and stay connected - in the esp-thread-br context.

I have been following the documentation in the OpenThread.io website to build the SW.

ESP-SDK: 

```
482a8fb2d7 (HEAD, tag: v5.1.2) change(version): Update version to 5.1.2
```
Example version v1.0:

```
b96fbd8 (HEAD, tag: v1.0) Merge branch 'feature/br_board_usb_init_refactor' into 'main'
```

### Actual behavior (suspected bug)

The Wi-Fi connection does not stay up, Wi-Fi is in state disconnected.
When trying to connect manually with wifi connect - the device reboots.

Network SSID and passphrase have been provided via the configuration.

### Error logs or terminal output

```shell
ELF file SHA256: 57c13efd8b5d9fdf

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x403758bf
0x403758bf: esp_restart_noos_dig at /home/jannek/openThread/esp32/esp-idf/components/esp_system/port/esp_system_chip.c:57 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1750
load:0x403c9700,len:0x4
load:0x403c9704,len:0xbe4
load:0x403cc700,len:0x2d18
entry 0x403c9908
I (26) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (26) boot: compile time Sep  9 2024 12:13:48
I (26) boot: Multicore bootloader
I (29) boot: chip revision: v0.2
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (78) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (86) boot:  3 ota_0            OTA app          00 10 00020000 00190000
I (93) boot:  4 ota_1            OTA app          00 11 001b0000 00190000
I (101) boot:  5 web_storage      Unknown data     01 82 00340000 00019000
I (108) boot:  6 rcp_fw           Unknown data     01 82 00359000 000a0000
[... 395 lines cut ...]
### Project release version

v1.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 22.04

### Shell

Bash

### Additional context

Wi-Fi network is based on TP-Link Deco M5, using the guest network.

Network is in IP-address range: 192.168.178.00/24

gytxxsy, 2024-09-12:
It seems that the device connected to Wi-Fi successfully and get ipv4 address `192.168.178.56`.

For command `wifi state` returns `disconnected` at `auto start mode`, it is a legacy issue present in `esp-thread-br v1.0`, which has been resolved in the latest code. You can switch to `esp-idf v5.1.3` and `esp-thread-br main`, and try again.

Additionally, when manually connecting to Wi-Fi, the SSID must be specified. For example, use `wifi connect -s my_ssid -p my_password` instead of `wifi connect`.

JanneKiiskila, 2024-09-12:
Why would you not then update the documentation to match this? Though it's probably not good to point to a floating main, but you should then advice:
- People to use ESP-IDF version v5.1.3 (instead of v5.1.2)
- Tag a new release v1.0.1 or similar from main and point people to it.

JanneKiiskila, 2024-09-12:
I can confirm Wi-Fi now stays online. However, the NRF 52840 is now not able to connect fully to the OpenThread network. Earlier was able.

```
<inf> net_state: Waiting for OT connection
```

Is the place where it hangs in the NRF 52840.
If I run discover in the ESP32 it says:

```
> discover

| Network Name     | Extended PAN     | PAN  | MAC Address      | Ch | dBm | LQI |
+------------------+------------------+------+------------------+----+-----+-----+
| OpenThread-7491  | 590a87a6774a99a3 | 7491 | d69404908b004a00 | 14 | -60 |   9 |
I(1612049) OPENTHREAD:[N] MeshForwarder-: Failed to send IPv6 UDP msg, len:56, chksum:c694, ecn:no, to:None, sec:no, error:Abort, prio:net
I(1612049) OPENTHREAD:[N] MeshForwarder-:     src:[fe80:0:0:0:d8d5:c426:8c38:9ece]:19788
I(1612049) OPENTHREAD:[N] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:2]:19788
Done
```

With the version 1.0 it did not give those errors of not being able to send.

JanneKiiskila, 2024-09-12:
There is still something funny going on with the Wi-Fi  or connectivity overall as well, as I would assume I should be able to do for example `dns resolve www.google.com` w/o issues. 

It does not work, it times out.

```
> dns resolve www.google.com

DNS response for www.google.com. - 
Error 28: ResponseTimeout

```

Ping does not seem to accept anything but direct IP-addresses.

gytxxsy, 2024-09-12:
In the documentation, we default to using the latest esp-thread-br commit. In fact, if you follow the `git clone xxx` command in the document, esp-idf will be on v5.1.3 and esp-thread-br will be on the main branch by default, with no need to manually switch. 

For the `esp-thread-br tag`, we will specify the corresponding esp-idf tag at the time of release. Please refer to:
https://github.com/espressif/esp-thread-br/releases/tag/v1.0

zwx1995esp, 2024-09-12:
> There is still something funny going on with the Wi-Fi or connectivity overall as well, as I would assume I should be able to do for example `dns resolve www.google.com` w/o issues.
> 
> It does not work, it times out.
> 
> ```
> > dns resolve www.google.com
> 
> DNS response for www.google.com. - 
> Error 28: ResponseTimeout
> ```
> 
> Ping does not seem to accept anything but direct IP-addresses.
> 
> ```
> > ping 142.251.116.94
> 
> Pinging synthesized IPv6 address: fdc5:5592:9dc2:2:0:0:8efb:745e
> 1 packets transmitted, 0 packets received. Packet loss = 100.0%.
> Done
> ```
> 
> ```
> > ipaddr
> 
> fd16:e34a:a433:eea8:0:ff:fe00:fc37
> fd16:e34a:a433:eea8:0:ff:fe00:fc10
> fdc5:5592:9dc2:1:b1e1:43a3:9ad6:b3d9
> fd16:e34a:a433:eea8:0:ff:fe00:fc00
> fd16:e34a:a433:eea8:0:ff:fe00:c000
> fd16:e34a:a433:eea8:55ed:c28b:cb21:f776
> fe80:0:0:0:d8d5:c426:8c38:9ece
> Done
> > netdata show
> 
> Prefixes:
> fdc5:5592:9dc2:1::/64 paros low c000
> Routes:
> fdc5:5592:9dc2:2:0:0::/96 sn low c000
> ::/0 sa med c000
> Services:
> 44970 5d fd16e34aa433eea855edc28bcb21f776d11f s c000
> Contexts:
> fdc5:5592:9dc2:1::/64 1 c
> Commissioning:
> 49983 c000 - 00000000000000000000000000000000
> Done
> > ping 8.8.8.8
> 
> Pinging synthesized IPv6 address: fdc5:5592:9dc2:2:0:0:808:808
> 1 packets transmitted, 0 packets received. Packet loss = 100.0%.
> Done
> ```

Hi @JanneKiiskila Please refer to [3.3. Service Discovery](https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/service_discovery.html#service-discovery), Notice a suffix `.default.service.arpa.` is needed, and also, the `dns resolve` should only work for the thread device(not BR but the other device in the thread network.) For the BR, it can access the wifi link, so if you want to resolve some websites, you can use the LWIP api directly.

For the ping command, also only works for the thread device. Hope this [FAQ](https://docs.espressif.com/projects/esp-thread-br/en/latest/qa.html) can help you a lot.

chshu, 2024-09-19:
@JanneKiiskila Hope the previous answers helped, feel free to reopen if you have any follow up questions.

closed 2024-09-19 by chshu