espressif/esp-thread-br #94: esp-thread-br not building against latest esp-idf (v3.5.1)

JanneKiiskila, 2024-09-11:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I would expect I should be able to update the ESF-IDF to latest version (to get latest openthread and Wi-Fi SW) against the `master` of this repository.

That's however not the case.

`esp-idf/examples/openthread/ot_rcp` was built succesfully.

esp-idf version:
```
commit c8fc5f643b7a7b0d3b182d3df610844e3dc9bd74 (HEAD, tag: v5.3.1)
Author: Marius Vikhammer <marius.vikhammer@espressif.com>
Date:   Tue Sep 3 09:37:24 2024 +0800

    change(version): Update version to 5.3.1

```

esp-thread-br version:
```
commit ceeb84afc77d868f786c92df93204fc5b0fe90cb (HEAD -> main, origin/main, origin/HEAD)
Merge: 6011f07 93716d9
Author: Shu Chen <chenshu@espressif.com>
Date:   Tue Sep 10 16:24:16 2024 +0800

    Merge branch 'fix/fix_the_rcp_update_issue' into 'main'
    
    fix(rcp update): fix a compilation issue
    
    See merge request espressif/esp-thread-br!138
```

### Actual behavior (suspected bug)

The build will fail with a complaint that the LwIP IPv6 addresses are not configured right.

### Error logs or terminal output

```shell
13:47 jannek@jannek-P720:~/openThread/esp32/esp-thread-br/examples/basic_thread_border_router$ idf.py fullclean
Executing action: fullclean
Executing action: remove_managed_components
Done
 14:53 jannek@jannek-P720:~/openThread/esp32/esp-thread-br/examples/basic_thread_border_router$ idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/jannek/openThread/esp32/esp-thread-br/examples/basic_thread_border_router/build' is empty. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig will be created.
Running cmake in directory /home/jannek/openThread/esp32/esp-thread-br/examples/basic_thread_border_router/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/jannek/.espressif/python_env/idf5.3_py3.12_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /home/jannek/openThread/esp32/esp-thread-br/examples/basic_thread_border_router"...
-- Existing sdkconfig '/home/jannek/openThread/esp32/esp-thread-br/examples/basic_thread_border_router/sdkconfig' renamed to '/home/jannek/openThread/esp32/esp-thread-br/examples/basic_thread_border_router/sdkconfig.old'.
-- Found Git: /usr/bin/git (found version "2.46.0") 
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/jannek/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/jannek/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc - skipped
[... 115 lines cut ...]
./install.sh
. ./export.sh
cd ..
git clone -b v1.0 --recursive https://github.com/espressif/esp-thread-br.git
```

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 22.04

### Shell

Bash

JanneKiiskila, 2024-09-11:
Seems the rcp is not compatible somehow?

gytxxsy, 2024-09-12:
It seems that the ESP32-H2 has an older version of the ot_rcp, which could lead to compatibility issues. 

You need to connect the ESP32-S3 and run `idf.py -p <port> erase-flash`, then connect the ESP32-H2 and run `idf.py -p <port> erase-flash`. After that, re-flash the latest firmware(based on `idf v5.3.1` and `esp-thread-br ceeb84af`) you built to the ESP32-S3, wait for the the `RCP auto update`.

JanneKiiskila, 2024-09-17:
> , then connect the ESP32-H2 and run `idf.py -p <port> erase-flash`. After that, re-flash the latest firmware(based on `idf v5.3.1` and `esp-thread-br ceeb84af`) you built to the ESP32-S3, wait for the the `RCP auto update`.

So, you mean that because I have once built this example with an earlier version - the firmware in the ESP32-H2 does not get updated at all with steps in the current documentation? The instructions are lacking this information then completely.

mys721tx, 2024-09-17:
Manually flashing the ot_rcp firmware (espressif/esp-idf@4d0db70) to esp32h2 and then the boarder router firmware (2eab508) to esp32s3 indeed fix the problem on my board. The number of IP addresses also requires manual fixing every time I switch targets.

JanneKiiskila, 2024-09-18:
Yes, I noticed the same. If you change the target you will lose all configuration changes. Not so pleasant or polite to override all configs without a warning.

chshu, 2024-09-19:
@mys721tx @JanneKiiskila If you are using the IDF v5.3.1 or later releases, you can change the default configuration here: https://github.com/espressif/esp-thread-br/blob/main/examples/basic_thread_border_router/sdkconfig.defaults#L63, then the configuration will be persistent.

The current IDF version suggested in the [Thread BR docs](https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#set-up-the-repositories) is v5.1.3, with which the default configuration works. We will update both the docs and default configuration to work with IDF v5.3.1 soon.

closed 2024-09-19 by chshu