espressif/esp-thread-br #95: Failed Building (menuconfig) on Windows: "Invalid character escape '\E'"

FutureCow, 2024-09-14:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Getting into menuconfig to setup the build

### Actual behavior (suspected bug)

get an error before getting into the setup menu

### Error logs or terminal output

```shell
C:\Espressif\frameworks\esp-thread-br-main\examples\basic_thread_border_router>idf.py menuconfig
Executing action: menuconfig
Running cmake in directory C:\Espressif\frameworks\esp-thread-br-main\examples\basic_thread_border_router\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\Espressif\frameworks\esp-thread-br-main\examples\basic_thread_border_router"...
-- IDF_TARGET is not set, guessed 'esp32s3' from sdkconfig 'C:/Espressif/frameworks/esp-thread-br-main/examples/basic_thread_border_router/sdkconfig.defaults'
-- Found Git: C:/Espressif/tools/idf-git/2.44.0/cmd/git.exe (found version "2.44.0.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Dependencies lock doesn't exist, solving dependencies.
Using component placed at C:\Espressif\frameworks\esp-thread-br-main\components\esp_ot_cli_extension for dependency "espressif/esp_ot_cli_extension", specified in C:\Espressif\frameworks\esp-thread-br-main\examples\basic_thread_border_router\main\idf_component.yml
Using component placed at C:\Espressif\frameworks\esp-thread-br-main\components\esp_rcp_update for dependency "espressif/esp_rcp_update", specified in C:\Espressif\frameworks\esp-thread-br-main\examples\basic_thread_border_router\main\idf_component.yml
..........NOTICE: Updating lock file at C:\Espressif\frameworks\esp-thread-br-main\examples\basic_thread_border_router\dependencies.lock
Processing 6 dependencies:
[1/6] espressif/esp-serial-flasher (0.0.11)
[2/6] espressif/esp_ot_cli_extension (1.2.3) (C:\Espressif\frameworks\esp-thread-br-main\components\esp_ot_cli_extension)
[3/6] espressif/esp_rcp_update (1.3.1) (C:\Espressif\frameworks\esp-thread-br-main\components\esp_rcp_update)
[4/6] espressif/iperf (0.1.3)
[5/6] espressif/mdns (1.4.0)
[6/6] idf (5.3.1)
-- Project sdkconfig file C:/Espressif/frameworks/esp-thread-br-main/examples/basic_thread_border_router/sdkconfig
Loading defaults file C:/Espressif/frameworks/esp-thread-br-main/examples/basic_thread_border_router/sdkconfig.defaults...
-- Compiler supported targets: xtensa-esp-elf
-- Found Python3: C:/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe (found version "3.11.2") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "esp_ot_br" version: 1
-- Adding linker script C:/Espressif/frameworks/esp-thread-br-main/examples/basic_thread_border_router/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Espressif/frameworks/esp-thread-br-main/examples/basic_thread_border_router/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.3.1/components/soc/esp32s3/ld/esp32s3.peripherals.ld
CMake Error at C:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/component.cmake:311 (foreach):
  Syntax error in cmake code at

    C:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/component.cmake:311

  when parsing string

    include;C:\Espressif\frameworks\esp-idf-v5.3.1/examples/common_components/protocol_examples_common/include

  Invalid character escape '\E'.
Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.3.1/tools/cmake/component.cmake:482 (__component_add_include_dirs)
  C:/Espressif/frameworks/esp-thread-br-main/components/esp_ot_cli_extension/CMakeLists.txt:34 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "C:/Espressif/frameworks/esp-thread-br-main/examples/basic_thread_border_router/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

### Steps to reproduce the behavior

Following the steps from: https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html

### Project release version

v5.3.1 ESP IDF with esp-thread-br-main

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

CMD

elgerg, 2024-09-16:
I get this on Windows 11 as well. I had to switch to Linux to get it to compile.

I was going to say work but this is the ticket I had to raise after I got compilations working:
https://github.com/espressif/esp-thread-br/issues/96

SMCorey, 2024-09-19:
I'm getting the same issue with the same setup.

I will try to do it on Linux tomorrow.

chshu, 2024-09-20:
@elgerg @SMCorey The compile issue on Windows should be fixed by https://github.com/espressif/esp-thread-br/commit/1318595bc390ddc66871174ffcd9bd0f29b797d0, could you try again? The change has been uploaded to the component managemen, tou don't need to download the latest code base, just clean the build history and re-build again.

SMCorey, 2024-09-20:
It worked. Thank you

closed 2024-09-23 by chshu