espressif/esp-thread-br #97: Get interface statistic

motters, 2024-09-17:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

It would be nice to be able to obtain interface statistics to monitor data usage.

Global stats for all interfaces seem to be available at `lwip/stats.h` easily enough. However it combines all interfaces (OpenThread, Wifi, Ethernet etc).

It would be better if these stats were accessible for each interface. These seem to be available through via `stats_mib2_netif_ctrs` at:
`esp_netif_get_handle_from_ifkey("WIFI_STA_DEF")->lwip_netif->mib2_counters`

However i cannot access this struct without getting compiler errors:
`error: invalid use of incomplete type 'esp_netif_t' {aka 'struct esp_netif_obj'}`

With `stats_mib2_netif_ctrs`, `lwip/stats.h` and `OpenThread History Functions` it would be possible to  gain useful statistics on each interface.  

### Use cases

This would help when the border router is using a cellular interface. Allowing you to keep track for any data increases and where the increase came from.

It also gives a deep in sight into the use of the OpenThread network and backbone.

### Alternatives

`lwip/stats.h` and `OpenThread History Functions` are available but they don't give all the information such as `total number of octets received on the interface` which `stats_mib2_netif_ctrs` does.

### Additional context

How can we access the `stats_mib2_netif_ctrs` for each interface?

motters, 2024-09-17:
Seem you can access netif via 
`auto if = (struct netif*) esp_netif_get_netif_impl(network::wifi::get_netif());`

However i'm not sure how to enable `MIB2_STATS`

zwx1995esp, 2024-09-18:
Hi @motters, thanks for addressing this. So just to confirm, for now, you only want to use the lwIP `mib_stats` feature, correct?

motters, 2024-09-18:
Hi,

Yes.

How can we access mib_stats feature (struct stats_mib2_netif_ctrs) for each network interface (WiFi, Ethernet, OpenThread)? Then we can wrap some REST APIs around it.

Cheers!

zwx1995esp, 2024-09-26:
Hi, just ask the question `how to enable the feature mib_stats in lwip`, and here are the replies from who is responsable to LwIP:
For now, we do not have a menuconfigration to enable this feature via `idf.py menuconfig`. And if you want to use this feature, you can refer to here: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/lwip.html#customized-lwip-options-from-esp-idf-build-system.
I have tried this, and it might work with these two changes:
1. add these two lines in the file `components/lwip/CMakeLists.txt`
```
    idf_component_get_property(lwip lwip COMPONENT_LIB)
    target_compile_definitions(${lwip} PRIVATE "-DMIB2_STATS=1")
```
2. enable the `LWIP_STATS` via `idf.py menuconfig`
[image]

then build your apps.

Please have a try. The LwIP group will support to enable this feature via `idf.py menuconfig` later.

motters, 2024-09-27:
Thanks @zwx1995esp I'll give this a try and report back what information can be obtained.

motters, 2024-10-31:
@zwx1995esp Sorry slow on this. I've tried the about bit still cannot access.

I cannot access the lwip_netif struct
`
error: invalid use of incomplete type 'esp_netif_t' {aka 'struct esp_netif_obj'}
   56 |                 auto wifi_data = esp_netif_get_handle_from_ifkey("WIFI_STA_DEF")->lwip_netif;
`

I cannot include the struct esp_netif_t as it's defined with esp_netif_lwip_internal.h which is not accessible.

zwx1995esp, 2024-11-01:
Hi, @motters  try this:
```
netif *wifi_data = esp_netif_get_netif_impl(esp_netif_get_handle_from_ifkey("WIFI_STA_DEF"));
```

motters, 2024-11-01:
@zwx1995esp Thanks thats works `netif *wifi_data = (struct netif*) esp_netif_get_netif_impl(esp_netif_get_handle_from_ifkey("WIFI_STA_DEF"));`

However to gain access to `wifi_data->mib2_counters` i had to open `esp/v5.1/esp-idf/components/lwip/lwip/src/include/lwip/opt.h` and add `#define MIB2_STATS 1`to line `2251` 

The code now compiles however the values within mib2_counters are always zero no matter how much data is transferred via the interface.

zwx1995esp, 2024-11-04:
> The code now compiles however the values within mib2_counters are always zero no matter how much data is transferred via the interface.

Hi @motters , this seems to be related to the lwIP components. you can create an issue in ESP-IDF to ask for some help from the person responsible for lwIP components.

chshu, 2024-11-29:
@motters please follow up in the IDF lwip component, feel free to link the new issue here.

closed 2024-11-29 by chshu