espressif/esp-thread-br #98: MBEDTLS ERROR ON build

platini76, 2024-09-19:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

it stops compiling after this error

### Actual behavior (suspected bug)

compile and run

### Error logs or terminal output

```shell
C:/Espressif/frameworks/esp-idf-v5.3.1/components/openthread/openthread/src/core/meshcop/secure_transport.cpp: In member function 'ot::Error ot::MeshCoP::SecureTransport::Setup(bool)':
C:/Espressif/frameworks/esp-idf-v5.3.1/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:353:5: error: 'mbedtls_ssl_conf_handshake_timeout' was not declared in this scope; did you mean 'mbedtls_ssl_conf_read_timeout'?
  353 |     mbedtls_ssl_conf_handshake_timeout(&mConf, 8000, 60000);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     mbedtls_ssl_conf_read_timeout
C:/Espressif/frameworks/esp-idf-v5.3.1/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:362:9: error: 'mbedtls_ssl_conf_dtls_cookies' was not declared in this scope; did you mean 'mbedtls_ssl_conf_dn_hints'?
  362 |         mbedtls_ssl_conf_dtls_cookies(&mConf, mbedtls_ssl_cookie_write, mbedtls_ssl_cookie_check, &mCookieCtx);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         mbedtls_ssl_conf_dn_hints
C:/Espressif/frameworks/esp-idf-v5.3.1/components/openthread/openthread/src/core/meshcop/secure_transport.cpp:378:16: error: 'mbedtls_ssl_set_hs_ecjpake_password' was not declared in this scope
  378 |         rval = mbedtls_ssl_set_hs_ecjpake_password(&mSsl, mPsk, mPskLength);
```

### Steps to reproduce the behavior

cd $IDF_PATH/examples/openthread/ot_rcp
idf.py set-target esp32h2
idf.py menuconfig
idf.py build
cd esp-thread-br/examples/basic_thread_border_router
idf.py set-target ESP32-S3
idf.py menuconfig
idf.py -p ${PORT_TO_BR} flash monitor

### Project release version

1.00

### System architecture

other (details in Additional context)

### Operating system

Windows

### Operating system version

windows 11

### Shell

PowerShell

### Additional context

try to flash a esp thread border router zigbee gateway v1.2

platini76, 2024-09-19:
now after some mods the error change at the end of building:

router_launch.c.obj -MF esp-idf\thread_border_router\CMakeFiles\__idf_thread_border_router.dir\src\border_router_launch.c.obj.d -o esp-idf/thread_border_router/CMakeFiles/__idf_thread_border_router.dir/src/border_router_launch.c.obj -c C:/Espressif/frameworks/esp-idf-v5.3.1/esp-thread-br/examples/common/thread_border_router/src/border_router_launch.c
C:/Espressif/frameworks/esp-idf-v5.3.1/esp-thread-br/examples/common/thread_border_router/src/border_router_launch.c: In function 'ot_br_init':
C:/Espressif/frameworks/esp-idf-v5.3.1/esp-thread-br/examples/common/thread_border_router/src/border_router_launch.c:119:9: error: implicit declaration of function 'esp_ot_wifi_config_get_ssid' [-Werror=implicit-function-declaration]
  119 |     if (esp_ot_wifi_config_get_ssid(wifi_ssid) == ESP_OK) {
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v5.3.1/esp-thread-br/examples/common/thread_border_router/src/border_router_launch.c:121:9: error: implicit declaration of function 'esp_ot_wifi_config_get_password' [-Werror=implicit-function-declaration]
  121 |         esp_ot_wifi_config_get_password(wifi_password);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v5.3.1/esp-thread-br/examples/common/thread_border_router/src/border_router_launch.c:127:9: error: implicit declaration of function 'esp_ot_wifi_connect'; did you mean 'esp_wifi_connect'? [-Werror=implicit-function-declaration]
  127 |     if (esp_ot_wifi_connect(wifi_ssid, wifi_password) == ESP_OK) {
      |         ^~~~~~~~~~~~~~~~~~~
      |         esp_wifi_connect
C:/Espressif/frameworks/esp-idf-v5.3.1/esp-thread-br/examples/common/thread_border_router/src/border_router_launch.c:142:9: error: implicit declaration of function 'esp_ot_wifi_border_router_init_flag_set' [-Werror=implicit-function-declaration]
  142 |         esp_ot_wifi_border_router_init_flag_set(true);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Espressif\frameworks\esp-idf-v5.3.1\esp-thread-br\examples\basic_thread_border_router\build\log\idf_py_stderr_output_8760 and C:\Espressif\frameworks\esp-idf-v5.3.1\esp-thread-br\examples\basic_thread_border_router\build\log\idf_py_stdout_output_8760

chshu, 2024-09-19:
@platini76 We don't see the issue with default configuration, could you share the commit ID of both esp-idf and esp-thread-br, and the configurations you have changed.

platini76, 2024-09-19:
and now another error: In file included from C:/Espressif/frameworks/esp-idf-v5.3.1/esp-thread-br/components/esp_ot_cli_extension/src/esp_ot_heap_diag.c:7:
C:/Espressif/frameworks/esp-idf-v5.3.1/esp-thread-br/components/esp_ot_cli_extension/include/esp_ot_heap_diag.h:11:10: fatal error: openthread/error.h: No such file or directory
   11 | #include <openthread/error.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
Missing "esp_openthread.h" file name found in the following component(s): openthread(C:\Espressif\frameworks\esp-idf-v5.3.1\components\openthread\include\esp_openthread.h). Maybe one of the components needs to add the missing header directory to INCLUDE_DIRS of idf_component_register call in CMakeLists.txt.
ninja failed with exit code 1, output of the command is in the C:\Espressif\frameworks\esp-idf-v5.3.1\esp-thread-br\examples\basic_thread_border_router\build\log\idf_py_stderr_output_11580 and C:\Espressif\frameworks\esp-idf-v5.3.1\esp-thread-br\examples\basic_thread_border_router\build\log\idf_py_stdout_output_11580

@chshu thanks for help, but what is a commit id?
I can share a photo:  
[image]

I am trying to follow these instructions: https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#build-and-run-the-thread-border-router

chshu, 2024-09-23:
@platini76 It seems that you didn't setup the IDF environment correctly, could you start with a clean environment and try again following the instructions here: https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#set-up-the-repositories.

chshu, 2024-11-29:
@platini76 Closing the issue, feel free to reopen if any follow up questions.

closed 2024-11-29 by chshu