espressif/esp-thread-br #105: Cannot use WebSocket Client due to constant disconnects

simondean, 2024-09-30:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

WebSocket client's connection stays open for multiple minutes and is reused for sending multiple WebSocket messages

### Actual behavior (suspected bug)

Every few seconds the client connection disconnects with this error: 

```
E (83821) websocket_client: esp_transport_connect() failed with -1, transport_error=ESP_OK, tls_error_code=0, tls_flags=0, esp_ws_handshake_status_code=101, errno=119
I (83831) esp_ot_br: WEBSOCKET_EVENT_ERROR
E (83831) esp_ot_br: Last error HTTP status code: 0x65
E (83831) esp_ot_br: Last error reported from tls stack: 0xb33fffff
I (83841) websocket_client: Reconnect after 10000 ms
I (83841) esp_ot_br: WEBSOCKET_EVENT_DISCONNECTED
E (83841) esp_ot_br: Last error HTTP status code: 0x65
```

### Error logs or terminal output

```shell
Detecting chip type... ESP32-S3
Running idf_monitor in directory **redacted**/esp/esp-thread-br/examples/basic_thread_border_router
Executing "**redacted**/.espressif/python_env/idf5.4_py3.12_env/bin/python **redacted**/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbmodem**redacted** -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 --revision 0 **redacted**/esp/esp-thread-br/examples/basic_thread_border_router/build/esp_ot_br.elf -m '**redacted**/.espressif/python_env/idf5.4_py3.12_env/bin/python' '**redacted**/esp/esp-idf/tools/idf.py'"...
--- esp-idf-monitor 1.5.0 on /dev/cu.usbmodem**redacted** 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048de8
--- 0x40048de8: uart_rx_readbuff in ROM

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1874
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc80
load:0x403cb700,len:0x2eb0
entry 0x403c8914
I (26) boot: ESP-IDF v5.4-dev-3201-g46acfdce96 2nd stage bootloader
I (27) boot: compile time Sep 29 2024 10:50:35
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (27) boot: efuse block revision: v1.3
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (28) boot.esp32s3: SPI Mode       : DIO
I (28) boot.esp32s3: SPI Flash Size : 4MB
I (28) boot: Enabling RNG early entropy source...
I (28) boot: Partition Table:
I (29) boot: ## Label            Usage          Type ST Offset   Length
I (29) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (29) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (30) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (30) boot:  3 ota_0            OTA app          00 10 00020000 00190000
I (31) boot:  4 ota_1            OTA app          00 11 001b0000 00190000
I (31) boot:  5 web_storage      Unknown data     01 82 00340000 00019000
I (32) boot:  6 rcp_fw           Unknown data     01 82 00359000 000a0000
I (32) boot: End of partition table
I (32) esp_image: segment 0: paddr=00020020 vaddr=3c0d0020 size=4c638h (312888) map
I (88) esp_image: segment 1: paddr=0006c660 vaddr=3fc95800 size=0330ch ( 13068) load
[... 231 lines cut ...]
```

### Steps to reproduce the behavior

See https://github.com/espressif/esp-thread-br/pull/104 which replicates the issue.  I have also reproduced the issue in a normal codebase (not the Thread Boarded Route codebase) and confirmed the issue does not occur there

### Project release version

unreleased master

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

Sonoma

### Shell

Fish

chshu, 2024-10-08:
@simondean Could you remove this line: https://github.com/espressif/esp-thread-br/blob/main/examples/common/thread_border_router/src/border_router_launch.c#L171, and try again? The issue seems caused by the default netif configuration.

simondean, 2024-10-09:
Hi.  I've just checked and it still happens if I remove this line:

```
    ESP_ERROR_CHECK(esp_netif_set_default_netif(openthread_netif));
```

Log output with line removed (same output as with line in place): 

```
I (218429) esp_ot_br: WEBSOCKET_EVENT_CONNECTED
I (222289) esp_ot_br: WEBSOCKET_EVENT_DATA
I (222289) esp_ot_br: Received opcode=1
W (222289) esp_ot_br: Received=echo.websocket.events sponsored by Lob.com
W (222289) esp_ot_br: Total payload length=42, data_len=42, current payload offset=0

I (222309) esp_ot_br: WEBSOCKET_EVENT_DATA
I (222309) esp_ot_br: Received opcode=10
W (222309) esp_ot_br: Received=
W (222309) esp_ot_br: Total payload length=0, data_len=0, current payload offset=0

I (224439) esp_ot_br: Successfully sent message
I (234469) esp_ot_br: Successfully sent message
E (235299) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 54
E (235299) websocket_client: esp_transport_poll_read() returned -1, transport_error=ESP_OK, tls_error_code=0, tls_flags=0, errno=119
I (235299) esp_ot_br: WEBSOCKET_EVENT_ERROR
E (235299) esp_ot_br: Last error HTTP status code: 0x65
I (235299) websocket_client: Reconnect after 10000 ms
I (235309) esp_ot_br: WEBSOCKET_EVENT_DISCONNECTED
E (235309) esp_ot_br: Last error HTTP status code: 0x65
E (235309) esp_ot_br: Last error captured as transport's socket errno: 0x71
I (240319) OPENTHREAD: Received RIO
I (240319) OPENTHREAD: prefix FD84:C890:3B8C:: lifetime 1800
```

chshu, 2024-11-29:
@simondean Do you have any update on the issue?

The logs didn't indicate any issue related to the BR feature, but some websocket issue, you can double check it using a Wi-Fi only device.

closed 2025-03-06 by zwx1995esp