espressif/esp-thread-br #106: Issue with RCP OTA update

syam-ari, 2024-10-14:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

RCP OTA update should work without any fail.

### Actual behavior (suspected bug)

Evertytime  i am seeing below error

RCP_UPDATE: esp_rcp_update(247): Failed to connect to RCP``

### Error logs or terminal output

```shell
We make the setup with ESP32S3 and ESP32H2. Unable to update the RCP OTA update and same tried using https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway code and no luck.

S3 Pin configuration as follow.

(Top) → ESP Thread Border Router Example → Board Configuration
  Espressif IoT Development Framework Configuration
(7) Pin to RCP reset
(8) Pin to RCP boot
(4) Pin to RCP TX
(5) Pin to RCP RX

H2 pin configuration as follow

(Top) → OpenThread RCP Example
  Espressif IoT Development Framework Configuration
[*] Configure RCP UART pin manually
(4)     The number of RX pin
(5)     The number of TX pin

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759d9
--- 0x403759d9: esp_restart_noos at /home/prasad/syam/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc10
load:0x403cb700,len:0x2dc0
entry 0x403c8904
I (26) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (26) boot: compile time Oct 10 2024 15:27:03
I (26) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (27) boot.esp32s3: SPI Mode       : DIO
I (27) boot.esp32s3: SPI Flash Size : 4MB
I (27) boot: Enabling RNG early entropy source...
I (28) boot: Partition Table:
I (28) boot: ## Label            Usage          Type ST Offset   Length
I (28) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (28) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (29) boot:  2 phy_init         RF data          01 01 00011000 00001000

I (29) boot:  3 ota_0            OTA app          00 10 00020000 00190000
I (30) boot:  4 ota_1            OTA app          00 11 001b0000 00190000
I (30) boot:  5 web_storage      Unknown data     01 82 00340000 00019000
I (30) boot:  6 rcp_fw           Unknown data     01 82 00359000 000a0000
I (31) boot: End of partition table
I (31) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=4e7f0h (321520) map
I (89) esp_image: segment 1: paddr=0006e818 vaddr=3fc99500 size=01800h (  6144) load
I (91) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=f3714h (997140) map
I (270) esp_image: segment 3: paddr=0016373c vaddr=3fc9ad00 size=036c4h ( 14020) load
I (274) esp_image: segment 4: paddr=00166e08 vaddr=40374000 size=15498h ( 87192) load
I (302) boot: Loaded app from partition at offset 0x20000
I (302) boot: Disabling RNG early entropy source...
I (303) cpu_start: Multicore app
I (312) cpu_start: Pro cpu start user code
I (312) cpu_start: cpu freq: 160000000 Hz
I (312) app_init: Application information:
I (313) app_init: Project name:     esp_ot_br
I (313) app_init: App version:      v1.1-2-g34812fe
I (313) app_init: Compile time:     Oct 14 2024 10:11:34
I (313) app_init: ELF file SHA256:  e6c0d009f...
I (313) app_init: ESP-IDF:          v5.3.1
I (313) efuse_init: Min chip rev:     v0.0
I (314) efuse_init: Max chip rev:     v0.99 
I (314) efuse_init: Chip rev:         v0.2
I (314) heap_init: Initializing. RAM available for dynamic allocation:
I (314) heap_init: At 3FCAE020 len 0003B6F0 (237 KiB): RAM
I (315) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (315) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (315) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (317) spi_flash: detected chip: gd
I (317) spi_flash: flash io: dio
W (317) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (318) sleep: Configure to isolate all GPIO pins in sleep state
I (318) sleep: Enable automatic switching of GPIO sleep configuration
I (319) main_task: Started on CPU0
I (329) main_task: Calling app_main()
I (369) RCP_UPDATE: RCP: using update sequence 1
I (379) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (379) OPENTHREAD: spinel UART interface initialization completed
I (379) main_task: Returned from app_main()
W(2389) OPENTHREAD:[W] P-SpinelDrive-: Wait for response timeout
I (2389) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (2389) esp_ot_br: Internal RCP Version: openthread-esp32/c8fc5f643b-f32c18bc1; esp32h2;  2024-10-14 04:39:55 UTC

I (2389) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (2389) gpio: GPIO[7]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2389) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (4339) RCP_UPDATE: esp_rcp_update(247): Failed to connect to RCP
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759d9

ESP32-H2 side logs

Build:Nov  1 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x408460e0,len:0x12f0
load:0x4083cad0,len:0xe80
load:0x4083efd0,len:0x29a0
entry 0x4083cada
I (23) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (24) boot: compile time Oct 14 2024 10:10:00
I (25) boot: chip revision: v0.1
I (27) boot.esp32h2: SPI Speed      : 64MHz
I (31) boot.esp32h2: SPI Mode       : DIO
I (36) boot.esp32h2: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=42028020 size=02124h (  8484) map
I (95) esp_image: segment 1: paddr=0001214c vaddr=40800000 size=05ecch ( 24268) load
I (109) esp_image: segment 2: paddr=00018020 vaddr=42000020 size=23078h (143480) map
I (153) esp_image: segment 3: paddr=0003b0a0 vaddr=40805ecc size=04e20h ( 20000) load
I (161) esp_image: segment 4: paddr=0003fec8 vaddr=4080acf0 size=00ff0h (  4080) load
I (167) boot: Loaded app from partition at offset 0x10000
I (167) boot: Disabling RNG early entropy source...
�D ����b=���~�bW�cJ�j����C�#��� ��������� �������ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0x1 (POWERON),boot:0x4 (DOWNLOAD(USB/UART0))
waiting for download
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0x1 (POWERON),boot:0x4 (DOWNLOAD(USB/UART0))
waiting for download
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0x1 (POWERON),boot:0x4 (DOWNLOAD(USB/UART0))
waiting for download
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0x1 (POWERON),boot:0x4 (DOWNLOAD(USB/UART0))
waiting for download
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0x1 (POWERON),boot:0x4 (DOWNLOAD(USB/UART0))
waiting for download
```

### Steps to reproduce the behavior

1. Build and flash the ot_rcp example code into the esp32h2 
2. Build and flash the esp-thread-br code into the esp32s3.
3. Observe the logs.

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 22.04.4 LTS

### Shell

ZSH

### Additional context

.Please find my setup 
[image]

chshu, 2024-10-24:
The H2 log indicates the BOOT and REST pins are configured correctly, but the UART pins don't work correctly. Could you use the default RCP pins configuration, and connect the H2 TX and RX pins directly?

You can also follow this example to setup [ot_br](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br) with standalong devkits.

syam-ari, 2024-11-06:
The default TX and RX pins of H2 works well.

closed 2024-11-06 by syam-ari