espressif/esp-thread-br #107: Documentation incorrect for ESP Thread BR CLI

mkleef, 2024-10-28:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expected the CLI example build to flash to the ESP H2

### Actual behavior (suspected bug)

Following the instructions here at: https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#build-and-run-the-thread-cli-device

1. Step is missing to idf.py set-target esp32s3 as by following the instructions it throws a chip mismatch error. 
2. When you do set to esp32s3, it fails to flash with the same error as trying to flash the Border Router itself. 

Note: RCP was built under the same $IDF_PATH of version 5.3.1

### Error logs or terminal output

```shell
W(2507) OPENTHREAD:[W] P-SpinelDrive-: Wait for response timeout
W(4517) OPENTHREAD:[W] P-SpinelDrive-: Wait for response timeout
E(4517) OPENTHREAD:[C] P-SpinelDrive-: Failed to reset co-processor!
E(4527) OPENTHREAD:[C] Platform------: ResetCoprocessor() at spinel_driver.cpp:160: Failure

abort() was called at PC 0x42008362 on core 0
0x42008362: syscall_not_implemented_aborts at C:/Espressif/frameworks/esp-idf-v5.3.1/components/newlib/syscalls.c:27

Backtrace: 0x40375c82:0x3fcab110 0x4037c359:0x3fcab130 0x40383c39:0x3fcab150 0x42008362:0x3fcab1c0 0x42098525:0x3fcab1e0 0x4205b6f1:0x3fcab200 0x4205bafd:0x3fcab220 0x4203568b:0x3fcab240 0x420351b1:0x3fcab290 0x42033c55:0x3fcab2c0 0x4200ae30:0x3fcab2f0 0x4037cea9:0x3fcab3e0
0x40375c82: panic_abort at C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/panic.c:463
0x4037c359: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/port/esp_system_chip.c:92
0x40383c39: abort at C:/Espressif/frameworks/esp-idf-v5.3.1/components/newlib/abort.c:38
0x42008362: syscall_not_implemented_aborts at C:/Espressif/frameworks/esp-idf-v5.3.1/components/newlib/syscalls.c:27
0x42098525: exit at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/stdlib/exit.c:65
0x4205b6f1: ot::Spinel::SpinelDriver::ResetCoprocessor(bool) at C:/Espressif/frameworks/esp-idf-v5.3.1/components/openthread/openthread/src/lib/spinel/spinel_driver.cpp:160 (discriminator 2)
0x4205bafd: ot::Spinel::SpinelDriver::Init(ot::Spinel::SpinelInterface&, bool, unsigned char const*, unsigned char) at C:/Espressif/frameworks/esp-idf-v5.3.1/components/openthread/openthread/src/lib/spinel/spinel_driver.cpp:81
0x4203568b: esp_openthread_radio_init at C:/Espressif/frameworks/esp-idf-v5.3.1/components/openthread/src/port/esp_openthread_radio_spinel.cpp:91 (discriminator 1)
0x420351b1: esp_openthread_platform_init at C:/Espressif/frameworks/esp-idf-v5.3.1/components/openthread/src/esp_openthread_platform.cpp:149
0x42033c55: esp_openthread_init at C:/Espressif/frameworks/esp-idf-v5.3.1/components/openthread/src/esp_openthread.cpp:69
0x4200ae30: ot_task_worker at C:/Espressif/frameworks/esp-idf-v5.3.1/examples/openthread/ot_cli/main/esp_ot_cli.c:71
0x4037cea9: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.3.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

### Steps to reproduce the behavior

cd $IDF_PATH/examples/openthread/ot_cli
idf.py -p COM3 flash monitor

### Project release version

5.3.1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

CMD

zwx1995esp, 2024-10-28:
Hi, @mkleef , from your logs: it seems like spinel host was used on ESP32H2. But the default config should use a radio native on H2(There is 15.4 modem on it). I think you might enable some specific configurations via `idf.py menuconfig`. So could you please run `idf.py fullclean` and `idf.py set-target esp32h2` to build again?

mkleef, 2024-10-28:
I realized what I did wrong. I did not understand that each USB-C port connected to a different half of the chipset. One being H2 and the other being S3. That should be made clearer in the documentation. But now that I've realized my error, I was able to flash H2 with the more up to date RCP, and get the S3 to work with either the BR or the CLI. Thanks for the response.

zwx1995esp, 2024-10-29:
Hi, @mkleef  I'm glad to hear you've resolved your issue!

As the documentation states: `Build the esp-idf/examples/openthread/ot_cli example and flash the firmware to another ESP32-H2 devkit.`

These are actually different devkit boards.

I'll close this issue for now, but feel free to reopen it if you encounter any further issues.

closed 2024-10-29 by zwx1995esp