espressif/esp-thread-br #109: espressif/esp_rcp_update v1.3.1 组件依赖 espressif/esp-serial-flasher ~0.0.0的 问题

wnylei, 2024-11-20:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

由于 v1.3.1版本的 espressif/esp_rcp_update  组件依赖了它 espressif/esp-serial-flasher ~0.0.0。 导致br项目无法支持 esp32c6  作为 rcp。 

**在这之前**

espressif/esp_rcp_update 的版本v1.0.2 ， 组件并没有添加依赖它 espressif/esp-serial-flasher。 br 项目中可以手动设置espressif/esp-serial-flasher 的版本为v1.3.0。  这样是可以是可以支持esp32c6的。

请问 ，如果我要使用v1.3.1版本的 espressif/esp_rcp_update 应该如何做，可以支持esp32c6 。

gytxxsy, 2024-11-25:
如果使用的是basic_thread_border_router这个example的话，可以直接将`esp-thread-br/components/esp_rcp_update/idf_component.yml`中的esp-serial-flasher修改为支持ESP32-C6的版本（例如1.3.0）。basic_thread_border_router目前在编译时会优先使用本地的esp_rcp_update组件。

如果使用的是一个其他repo的example的话，需要先将`esp-thread-br/components/esp_rcp_update/idf_component.yml`中的esp-serial-flasher修改为支持ESP32-C6的版本，然后再修改example的idf_component.yml，指定使用本地的esp_rcp_update，例如：
* example的原idf_component.yml：
```
espressif/esp_rcp_update:
    version: "~1.3.0"
```
* example的修改后的idf_component.yml：
```
espressif/esp_rcp_update:
   path: "<你的路径>/esp-thread-br/components/esp_rcp_update"
```

wnylei, 2024-11-29:
好的,多谢.  
目前测试:
espressif/esp-serial-flasher: version: 1.3.0 可以正常使用 esp32-c6 作为 rcp.
                                                 version: 1.3.1 是有问题的.

gytxxsy, 2025-03-25:
basic_thread_border_router的`esp_rcp_update`组件已经更新至1.4.0版本，支持使用esp32-c6作为rcp.