espressif/esp-thread-br #111: failure to build each time on MacOS and linux debian bookworm

zappydood, 2024-12-04:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

to compile and flash successfuly

### Actual behavior (suspected bug)

gives the same exact error on macOS as it does on linux, resulting in a fail to build and flash

### Error logs or terminal output

```shell
asshat@sonic:~/esp-thread-br/examples/basic_thread_border_router$ idf.py menuconfig
Executing action: menuconfig
Running ninja in directory /home/asshat/esp-thread-br/examples/basic_thread_border_router/build
Executing "ninja menuconfig"...
[0/1] cd /home/asshat/esp-thread-br/examples/basic_th...read-br/examples/basic_thread_border_router/sdkconfig
Loading defaults file /home/asshat/esp-thread-br/examples/basic_thread_border_router/sdkconfig.defaults...
TERM environment variable is set to "xterm-256color"
Loaded configuration '/home/asshat/esp-thread-br/examples/basic_thread_border_router/sdkconfig'
Configuration saved to '/home/asshat/esp-thread-br/examples/basic_thread_border_router/sdkconfig'
Loading defaults file /home/asshat/esp-thread-br/examples/basic_thread_border_router/sdkconfig.defaults...
asshat@sonic:~/esp-thread-br/examples/basic_thread_border_router$ idf.py -p /dev/ttyACM0 flash monitor
Executing action: flash
Running ninja in directory /home/asshat/esp-thread-br/examples/basic_thread_border_router/build
Executing "ninja flash"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32s3
Using component placed at /home/asshat/esp-thread-br/components/esp_br_http_ota for dependency "esp_br_http_ota", specified in /home/asshat/esp-thread-br/examples/basic_thread_border_router/main/idf_component.yml
Using component placed at /home/asshat/esp-thread-br/components/esp_ot_br_server for dependency "esp_ot_br_server", specified in /home/asshat/esp-thread-br/examples/basic_thread_border_router/main/idf_component.yml
Using component placed at /home/asshat/esp-thread-br/components/esp_ot_cli_extension for dependency "espressif/esp_ot_cli_extension", specified in /home/asshat/esp-thread-br/examples/basic_thread_border_router/main/idf_component.yml
Using component placed at /home/asshat/esp-thread-br/components/esp_rcp_update for dependency "espressif/esp_rcp_update", specified in /home/asshat/esp-thread-br/examples/basic_thread_border_router/main/idf_component.yml
Using component placed at /home/asshat/esp-idf/examples/common_components/protocol_examples_common for dependency "protocol_examples_common", specified in /home/asshat/esp-thread-br/examples/basic_thread_border_router/main/idf_component.yml
Using component placed at /home/asshat/esp-thread-br/examples/common/thread_border_router for dependency "thread_border_router", specified in /home/asshat/esp-thread-br/examples/basic_thread_border_router/main/idf_component.yml
Processing 10 dependencies:
[10/10] idf (5.3.1)
-- Project sdkconfig file /home/asshat/esp-thread-br/examples/basic_thread_border_router/sdkconfig
Loading defaults file /home/asshat/esp-thread-br/examples/basic_thread_border_router/sdkconfig.defaults...
-- Compiler supported targets: xtensa-esp-elf
-- App "esp_ot_br" version: v1.1-8-g17946c1
-- Adding linker script /home/asshat/esp-thread-br/examples/basic_thread_border_router/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/asshat/esp-thread-br/examples/basic_thread_border_router/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /home/asshat/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /home/asshat/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /home/asshat/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /home/asshat/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script /home/asshat/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /home/asshat/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /home/asshat/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_br_http_ota esp_coex esp_common esp_driver_ana_cmpr esp_driver_cam esp_driver_dac esp_driver_gpio esp_driver_gptimer esp_driver_i2c esp_driver_i2s esp_driver_isp esp_driver_jpeg esp_driver_ledc esp_driver_mcpwm esp_driver_parlio esp_driver_pcnt esp_driver_ppa esp_driver_rmt esp_driver_sdio esp_driver_sdm esp_driver_sdmmc esp_driver_sdspi esp_driver_spi esp_driver_touch_sens esp_driver_tsens esp_driver_uart esp_driver_usb_serial_jtag esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_ot_br_server esp_ot_cli_extension esp_partition esp_phy esp_pm esp_psram esp_rcp_update esp_ringbuf esp_rom esp_system esp_timer esp_vfs_console esp_wifi espcoredump espressif__esp-serial-flasher espressif__iperf espressif__mdns esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthread partition_table perfmon protobuf-c protocol_examples_common protocomm pthread sdmmc soc spi_flash spiffs tcp_transport thread_border_router touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/asshat/esp-idf/components/app_trace /home/asshat/esp-idf/components/app_update /home/asshat/esp-idf/components/bootloader /home/asshat/esp-idf/components/bootloader_support /home/asshat/esp-idf/components/bt /home/asshat/esp-idf/components/cmock /home/asshat/esp-idf/components/console /home/asshat/esp-idf/components/cxx /home/asshat/esp-idf/components/driver /home/asshat/esp-idf/components/efuse /home/asshat/esp-idf/components/esp-tls /home/asshat/esp-idf/components/esp_adc /home/asshat/esp-idf/components/esp_app_format /home/asshat/esp-idf/components/esp_bootloader_format /home/asshat/esp-thread-br/components/esp_br_http_ota /home/asshat/esp-idf/components/esp_coex /home/asshat/esp-idf/components/esp_common /home/asshat/esp-idf/components/esp_driver_ana_cmpr /home/asshat/esp-idf/components/esp_driver_cam /home/asshat/esp-idf/components/esp_driver_dac /home/asshat/esp-idf/components/esp_driver_gpio /home/asshat/esp-idf/components/esp_driver_gptimer /home/asshat/esp-idf/components/esp_driver_i2c /home/asshat/esp-idf/components/esp_driver_i2s /home/asshat/esp-idf/components/esp_driver_isp /home/asshat/esp-idf/components/esp_driver_jpeg /home/asshat/esp-idf/components/esp_driver_ledc /home/asshat/esp-idf/components/esp_driver_mcpwm /home/asshat/esp-idf/components/esp_driver_parlio /home/asshat/esp-idf/components/esp_driver_pcnt /home/asshat/esp-idf/components/esp_driver_ppa /home/asshat/esp-idf/components/esp_driver_rmt /home/asshat/esp-idf/components/esp_driver_sdio /home/asshat/esp-idf/components/esp_driver_sdm /home/asshat/esp-idf/components/esp_driver_sdmmc /home/asshat/esp-idf/components/esp_driver_sdspi /home/asshat/esp-idf/components/esp_driver_spi /home/asshat/esp-idf/components/esp_driver_touch_sens /home/asshat/esp-idf/components/esp_driver_tsens /home/asshat/esp-idf/components/esp_driver_uart /home/asshat/esp-idf/components/esp_driver_usb_serial_jtag /home/asshat/esp-idf/components/esp_eth /home/asshat/esp-idf/components/esp_event /home/asshat/esp-idf/components/esp_gdbstub /home/asshat/esp-idf/components/esp_hid /home/asshat/esp-idf/components/esp_http_client /home/asshat/esp-idf/components/esp_http_server /home/asshat/esp-idf/components/esp_https_ota /home/asshat/esp-idf/components/esp_https_server /home/asshat/esp-idf/components/esp_hw_support /home/asshat/esp-idf/components/esp_lcd /home/asshat/esp-idf/components/esp_local_ctrl /home/asshat/esp-idf/components/esp_mm /home/asshat/esp-idf/components/esp_netif /home/asshat/esp-idf/components/esp_netif_stack /home/asshat/esp-thread-br/components/esp_ot_br_server /home/asshat/esp-thread-br/components/esp_ot_cli_extension /home/asshat/esp-idf/components/esp_partition /home/asshat/esp-idf/components/esp_phy /home/asshat/esp-idf/components/esp_pm /home/asshat/esp-idf/components/e [... cut ...]
tried running this on m2 mac mini with apple silicon as well as debian bookworm on x86, on macOS zsh was used and on debian bash.

As I am new to using esp-idf, I am a bit confused and the guide doesn't give tons of details for each and every scenario. But I would like to get this fixed. I followed the guide exactly. I correctly input all the additional settings it asks for, using the correct versions etc. But its the same failure each time and I don't fully understand where I am going wrong. Help would be appreciated.

chshu, 2024-12-04:
There was a recent change [ccf5b28](https://github.com/espressif/esp-thread-br/commit/ccf5b28364913f2ff74c76b04ac488cd09cc3473) that needs the corresponding change in IDF: [f41b43d](https://github.com/espressif/esp-idf/commit/f41b43dc431f39a38c40e92cb6a6d31389cdedfe).

Please either checkout the latest IDF master branch, or revert the [ccf5b28](https://github.com/espressif/esp-thread-br/commit/ccf5b28364913f2ff74c76b04ac488cd09cc3473) change to work with old IDF releases.

FMistrorigo, 2024-12-04:
Hi,

I'm having the same issue, can you at least update the README.txt file?
It's really annoyng to setup esp-idf on version 5.3.1 only to find out it's not correct after building...

Thanks,
Fabio

chshu, 2024-12-06:
The docs is updated, sorry for the inconvenience.

closed 2024-12-06 by chshu