espressif/esp-thread-br #112: Upgrading from ESP-IDF 5.1 to 5.3; breaking changes?

motters, 2024-12-17:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hi,

### Summary

Are there any API changes that could break routing data from the NAT to the backbone when migrating from IDF 5.1 to 5.3?

### Details

We were having issues with the ESP BR crashing during high bursts of data, (https://github.com/orgs/openthread/discussions/11038). @chshu suggested we upgraded to IDF 5.3 which fixed a bug around this issue.

We have updated ESP-IDF to 5.3 and everything is working, except the BR is not passing requests from the NAT to the backbone.

For example, when a node tries to perform a DNS request using the BR's NAT, there is no reply. Below shows the BR log where `fd6e:266c:4da0:1:7d59:1874:de24:c55e` is the node.

```
I(727307) OPENTHREAD:[I] MeshForwarder-: Received IPv6 UDP msg, len:94, chksum:6b09, ecn:no, from:0x5c00, sec:yes, prio:normal, rss:-45.0
I(727320) OPENTHREAD:[I] MeshForwarder-:     src:[fd6e:266c:4da0:1:7d59:1874:de24:c55e]:41620
I(727322) OPENTHREAD:[I] MeshForwarder-:     dst:[fd6e:266c:4da0:2:0:0:808:808]:53
I(727338) OPENTHREAD:[I] Mac-----------: Frame rx failed, error:Duplicated, len:101, seqnum:227, type:Data, src:0x5c00, dst:0x3c00, sec:yes, ackreq:yes
I(727354) OPENTHREAD:[I] MeshForwarder-: Sent IPv6 UDP msg, len:199, chksum:fe68, ecn:no, to:0xffff, sec:yes, prio:net
I(727357) OPENTHREAD:[I] MeshForwarder-:     src:[fe80:0:0:0:2c81:3eaf:10a9:1e40]:19788
I(727369) OPENTHREAD:[I] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788
I(727408) OPENTHREAD:[I] MeshForwarder-: Received IPv6 UDP msg, len:199, chksum:1aea, ecn:no, from:42e2345cff43d991, sec:yes, prio:net, rss:-45.5
I(727421) OPENTHREAD:[I] MeshForwarder-:     src:[fe80:0:0:0:40e2:345c:ff43:d991]:19788
I(727423) OPENTHREAD:[I] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788
I(727436) OPENTHREAD:[I] Mle-----------: Receive Data Response (fe80:0:0:0:40e2:345c:ff43:d991)
I(728184) OPENTHREAD:[I] RoutingManager: Evaluating routing policy
I(728186) OPENTHREAD:[I] RoutingManager: Evaluating NAT64 prefix
I(728188) OPENTHREAD:[I] RoutingManager: Preparing RA
I(728199) OPENTHREAD:[I] RoutingManager: - RA Header - flags - M:0 O:0 S:1
I(728210) OPENTHREAD:[I] RoutingManager: - RA Header - default route - lifetime:0
I(728220) OPENTHREAD:[I] RoutingManager: - PIO fd00:0:0:0::/64 (valid:1800, preferred:1800)
I(728223) OPENTHREAD:[I] RoutingManager: - RIO fd6e:266c:4da0:1::/64 (lifetime:1800, prf:medium)
I(728237) OPENTHREAD:[I] RoutingManager: Sent RA on infra netif 2
I(728240) OPENTHREAD:[I] RoutingManager: Will evaluate routing policy in 03:02.308 (182308 msec)
I(728253) OPENTHREAD:[I] RoutingManager: Received RA from fe80:0:0:0:66b7:8ff:fe3d:d6ac on infra netif 2 (this BR routing-manager)
I(731700) OPENTHREAD:[I] Mle-----------: Send Advertisement (ff02:0:0:0:0:0:0:1)
I(731740) OPENTHREAD:[I] MeshForwarder-: Sent IPv6 UDP msg, len:91, chksum:d99d, ecn:no, to:0xffff, sec:no, prio:net
I(731742) OPENTHREAD:[I] MeshForwarder-:     src:[fe80:0:0:0:2c81:3eaf:10a9:1e40]:19788
I(731753) OPENTHREAD:[I] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788
I(741371) OPENTHREAD:[I] Mle-----------: Send Announce on channel 13
I(741399) OPENTHREAD:[I] MeshForwarder-: Sent IPv6 UDP msg, len:83, chksum:c9bf, ecn:no, to:0xffff, sec:yes, prio:net
I(741401) OPENTHREAD:[I] MeshForwarder-:     src:[fe80:0:0:0:2c81:3eaf:10a9:1e40]:19788
I(741413) OPENTHREAD:[I] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788
I(744228) OPENTHREAD:[I] Mle-----------: Send Advertisement (ff02:0:0:0:0:0:0:1)
I(744249) OPENTHREAD:[I] MeshForwarder-: Sent IPv6 UDP msg, len:91, chksum:d541, ecn:no, to:0xffff, sec:no, prio:net
I(744251) OPENTHREAD:[I] MeshForwarder-:     src:[fe80:0:0:0:2c81:3eaf:10a9:1e40]:19788
I(744263) OPENTHREAD:[I] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788
I(749409) OPENTHREAD:[I] MeshForwarder-: Received IPv6 UDP msg, len:91, chksum:382c, ecn:no, from:42e2345cff43d991, sec:no, prio:net, rss:-46.0
```

The BR can contact the node:
 
```
> ping fd6e:266c:4da0:1:7d59:1874:de24:c55e

I(868590) OPENTHREAD:[I] Icmp6---------: Sent echo request: (seq = 3)
I(868609) OPENTHREAD:[I] MeshForwarder-: Sent IPv6 ICMP6 msg, len:56, chksum:2217, ecn:no, to:0x5c00, sec:yes, prio:normal
I(868612) OPENTHREAD:[I] MeshForwarder-:     src:[fd6e:266c:4da0:1:167e:fa1f:76f2:7ec7]
I(868624) OPENTHREAD:[I] MeshForwarder-:     dst:[fd6e:266c:4da0:1:7d59:1874:de24:c55e]
I(868645) OPENTHREAD:[I] MeshForwarder-: Received IPv6 ICMP6 msg, len:56, chksum:2117, ecn:no, from:0x5c00, sec:yes, prio:normal, rss:-43.0
I(868658) OPENTHREAD:[I] MeshForwarder-:     src:[fd6e:266c:4da0:1:7d59:1874:de24:c55e]
I(868660) OPENTHREAD:[I] MeshForwarder-:     dst:[fd6e:266c:4da0:1:167e:fa1f:76f2:7ec7]
16 bytes from fd6e:266c:4da0:1:7d59:1874:de24:c55e: icmp_seq=3 hlim=64 time=85ms
1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 85/85.0/85 ms.
```

The BR's NAT is active:

```
> nat64 state

PrefixManager: Active
Done
```

The BR has an internet connection as we are running other libraries that use the same network. However OT cannot ping any external ip addresses:

```
> ping 8.8.8.8

Pinging synthesized IPv6 address: fd6e:266c:4da0:2:0:0:808:808
1 packets transmitted, 0 packets received. Packet loss = 100.0%.
Done
```

The nodes use OT SRP and these services are registered to the building's router through the BR successfully. Hence proving the backbone is registered correctly. See the below output from a computer on the same network.

```
user@mac folder %  dns-sd -B _XXX._udp
Browsing for _XXX._udp
DATE: ---Tue 17 Dec 2024---
10:41:41.122  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
10:41:41.122  Add        2  11 local.               _XXX._udp.    f4ce36dc0e4c1111
```

Some BR logs that may help:

```
> bbr stateI

Primary
Done

> br state

running
Done

> netstate

| Local Address                                   | Peer Address                                    |
+-------------------------------------------------+-------------------------------------------------+
| [0:0:0:0:0:0:0:0]:1000                          | [0:0:0:0:0:0:0:0]:0                             |
| [0:0:0:0:0:0:0:0]:0                             | [0:0:0:0:0:0:0:0]:0                             |
| [0:0:0:0:0:0:0:0]:49156                         | [fda7:56f4:942:ab5e:4490:7063:15ef:77bf]:53549  |
| [0:0:0:0:0:0:0:0]:53549                         | [0:0:0:0:0:0:0:0]:0                             |
| [0:0:0:0:0:0:0:0]:49153                         | [0:0:0:0:0:0:0:0]:0                             |
| [0:0:0:0:0:0:0:0]:53                            | [0:0:0:0:0:0:0:0]:0                             |
| [0:0:0:0:0:0:0:0]:61631                         | [0:0:0:0:0:0:0:0]:0                             |
| [0:0:0:0:0:0:0:0]:19788                         | [0:0:0:0:0:0:0:0]:0                             |
| [0:0:0:0:0:0:0:0]:61631                         | [0:0:0:0:0:0:0:0]:0                             |
Done

> router table

| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     | Link |
+----+--------+----------+-----------+-------+--------+-----+------------------+------+
| 15 | 0x3c00 |       63 |         0 |     0 |      0 |   0 | 0000000000000000 |    0 |
| 23 | 0x5c00 |       63 |         0 |     3 |      3 |   1 | 42e2345cff43d991 |    1 |

> netdata show

Prefixes:
fd6e:266c:4da0:1::/64 paos low 3c00
Routes:
fc00::/7 sa med 3c00
fd6e:266c:4da0:2:0:0::/96 sn low 3c00
Services:
44970 01 68000500000e10 s 3c00 0
44970 5d fda756f40942ab5e4490706315ef77bfd12d s 3c00 1
Contexts:
fd6e:266c:4da0:1::/64 1 c
Commissioning:
19608 3c00 - ff
Done

> domainname

DefaultDomain
Done

> ifconfig

up
Done

> ipaddr

fda7:56f4:942:ab5e:0:ff:fe00:fc30
fda7:56f4:942:ab5e:0:ff:fe00:fc11
fda7:56f4:942:ab5e:0:ff:fe00:fc38
fd6e:266c:4da0:1:167e:fa1f:76f2:7ec7
fda7:56f4:942:ab5e:0:ff:fe00:fc10
fda7:56f4:942:ab5e:0:ff:fe00:fc00
fda7:56f4:942:ab5e:0:ff:fe00:3c00
fda7:56f4:942:ab5e:4490:7063:15ef:77bf
fe80:0:0:0:2c81:3eaf:10a9:1e40
Done

> ipmaddr

ff33:40:fda7:56f4:942:ab5e:0:1
ff32:40:fda7:56f4:942:ab5e:0:1
ff02:0:0:0:0:0:0:2
ff03:0:0:0:0:0:0:2
ff02:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:fc
Done

> leaderdata

Partition ID: 1373084161
Weighting: 65
Data Version: 230
Stable Data Version: 103
Leader Router ID: 15
Done
``` 

Do you have any suggestion?

Thanks!

zwx1995esp, 2024-12-18:
Hi @motters did you use the tag: v5.3.2 for your test or could you please share the commit hash you used? I try to reproduce the issue you encountered but failed. All features related NAT64 works well on my side.

We found a problem on the tag v5.3.2 and we are investigating it these days.(There will be an error logs `Failed to get LL address, error: Invalid If index` during BR attaching and becoming router or leader. But even with this problem, I still can not reproduce the issues you encountered.)  

So could you please try the code base metioned in the docs: https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#set-up-the-repositories and see if NAT64 works?

For the dns resolving command you metioned above:
```
The BR's openthread stack can also not perform DNS requests

> dns resolve google.com 8.8.8.8

Synthesized IPv6 DNS server address: fd6e:266c:4da0:2:0:0:808:808
E (2195814) OPENTHREAD: Failed to Send UDP message, err: -12
E (2201814) OPENTHREAD: Failed to Send UDP message, err: -12
E (2207815) OPENTHREAD: Failed to Send UDP message, err: -12
DNS response for google.com. - 
Error 28: ResponseTimeout

```
Did you execute the command on the BR device or on the node?

motters, 2024-12-18:
HI @zwx1995esp thanks for the reply. We have tried both 5.3.1 & 5.3.2 releases. I will try the commit mentioned in the documentation this morning.

All commands were executed from the BR, including `dns resolve google.com 8.8.8.8`. Although i can confirm the node side has the same issue (timing out connections).

Maybe one difference is we have https://github.com/espressif/arduino-esp32 package installed to speed up development. Could this be causing issues?

zwx1995esp, 2024-12-18:
Hi, @motters  just a question, does this `dns resolve google.com 8.8.8.8` work on BR if using the previous code? Based our docs: [why-can-t-i-access-the-host-from-the-br-device-using-the-ping-command](https://docs.espressif.com/projects/esp-thread-br/en/latest/qa.html#q1-why-can-t-i-access-the-host-from-the-br-device-using-the-ping-command), `DNS` comand is also implemented on Thread layer, I don't think that BR can resovled any address using this command. If you want to performance DNS on BR, you need also implement and use the LWIP layer DNS resolving command.

zwx1995esp, 2024-12-18:
This DNS resolve command is used for the end node devices which attached in BR's thread network. So could you please also try if it works on the end node devices?

motters, 2024-12-18:
@zwx1995esp i see, is there a command i can run on the BR to test the network?

I'll test from a node and post the result.

motters, 2024-12-18:
Good news the dns command does work from the node when using the BR is using the `basic_thread_border_router` example,
[image]

However it does not work from our modded / production version on the BR. I suppose next step is to exclude idf-ardunio causing the issue? 
[image]

motters, 2024-12-19:
I'm comparing the logs of `basic_thread_border_router` to our production firmware we're upgrading. We're getting the following error which `basic_thread_border_router` does not. Any advise on what this could be?
[image]

We are also getting multiple of the following warnings:
[image]

However the `basic_thread_border_router` seems to get similar:
[image]

So i'm assuming the issues is todo with `Failed to probe backbone multicast listeners`

motters, 2024-12-19:
After more digging it seems the esp-arduino WebServer is the component causing the issues.

Once we remove all the WebSever code OpenThread is now working correctly.

motters, 2024-12-23:
@zwx1995esp @chshu I think this can be closed. It seems using any `arduino-esp32` networking features breaks the OT routing, assuming NAT64. 

arduino-esp32 `mDNS` and `webserver` library had to be removed and replaced with esp-idf libraries. As `arduino-esp32` is an official Espressif product, might be worth looking into why?

However, i think it's worth noting in the README that OpenThread BR features cannot be used in conjunction with `arduino-esp32` networking libraries. Hopefully this will save other people some time :)