espressif/esp-thread-br #113: ESP Thread BR in Wi-Fi AP mode

walker2398, 2024-12-17:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

As mentioned in this [post](https://yplam.com/IOT/openthread/raspberry-pi-openthread/),Raspberry pies can build a thread BR in its AP mode.Can ESP Thread BR make this work?

### Use cases

In my use case, I want to allow user to build a thread network with the ESP Thread BR  **only**,without any outer AP. So connecting to the board by its Wi-Fi AP mode is a must.

### Alternatives

Maybe using bluetooth communication might work?

chshu, 2024-12-18:
@walker2398 In your use case, will the ESP Thread BR connect to Ethernet? Or is it a purely local network where only Thread devices (connected to the BR's Thread network) and Wi-Fi devices (connected to the BR's AP) can communicate with each other, with no access from external devices?

walker2398, 2024-12-18:
@chshu 原谅我英语不好，只能写中文。
是的，我的Thread网络应该连接到外网，只是一个本地网络，手机通过连接Border Router控制网络内设备。

walker2398, 2025-03-10:
这太酷了！感谢你们的付出。

closed 2025-03-10 by walker2398