espressif/esp-thread-br #114: BR not auto connects to wifi if network is lost.

donoo, 2024-12-28:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

when wifi is lost due to some reason. Browser router does not come online. Need to manually restart to connect it back to wifi network. 

### Actual behavior (suspected bug)

It should automatically connect to wifi when network is again available. 

### Steps to reproduce the behavior

1. Start the Thread Broder router. 
2. Check if is available in WIFI network in your WIFI router. 
4. After that turn off the wifi router. 
5. Wait for 5-10 minutes
6. Turn on the wifi router. 
7. Check for the Thread Border router in the network. 

### Project release version

openthread-esp32/9d7f2d69f5-005c5cefc; esp32h2; 2024-12-22 15:13:57 UTC

### System architecture

other (details in Additional context)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

PowerShell

### Additional context

I have used following document to build the border router : https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/index.html

I m using single board mentioned [here](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html)

chshu, 2024-12-31:
@donoo Please refer to the logic here: [example_handler_on_wifi_disconnect](https://github.com/espressif/esp-idf/blob/master/examples/common_components/protocol_examples_common/wifi_connect.c#L157), and add a similar code in your application.

We will enhance the br example to support the wifi reconnection.

donoo, 2024-12-31:
@chshu  Thanks for you reply.  If possible, can you specify which file I need to update? 

I m newbee in esp32 stuff.

gytxxsy, 2025-03-25:
@donoo 
In the latest example (e.g., commit a5ed23e1a1abc8973c5f5c629de4cde44aef5b7c and later), we have enhanced the BR example, and it now supports Wi-Fi reconnection. The number of Wi-Fi reconnection attempts is limited and you can adjust the maximum retry count via `idf.py menuconfig → Example Connection Configuration → Maximum retry`.

closed 2025-03-25 by zwx1995esp