espressif/esp-thread-br #115: Cannot add Matter over thread device due to "mdns: Cannot allocate memory"

mrazekfilip, 2025-01-01:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

often

### Expected behavior

successfuly connected Matter over thread device to HomeAssistant via the esp thread border router (flashed on [esp thread border router board](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html)) 

### Actual behavior (suspected bug)

The device seems to be connected (the pairing led doesnt blink), but the device is not connected. 

### Error logs or terminal output

```shell
[0;32mI (9208664) OPENTHREAD: Multicast listener add: FF05::2[0m
[0;32mI (9211844) OPENTHREAD: Received RIO[0m
[0;32mI (9211844) OPENTHREAD: prefix FDEE:80F3:66D:1:: lifetime 1800
[0m
[0;32mI (9211854) OPENTHREAD: Multicast listener add: FF05::2[0m
[0;32mI (9212044) OPENTHREAD: Multicast listener add: FF05::2[0m
[0;32mI (9215424) OPENTHREAD: Received RIO[0m
[0;32mI (9215424) OPENTHREAD: prefix FDEE:80F3:66D:1:: lifetime 1800
[0m
[0;32mI (9215434) OPENTHREAD: Multicast listener add: FF05::2[0m
[0;32mI (9218494) OPENTHREAD: Received RIO[0m
[0;32mI (9218494) OPENTHREAD: prefix FDEE:80F3:66D:1:: lifetime 1800
[0m
[0;32mI (9218504) OPENTHREAD: Multicast listener add: FF05::2[0m
[0;32mI (9305854) OPENTHREAD: Host delete[0m
[0;32mI (9305854) OPENTHREAD: delete host 5E3EECCCE3B4EBD1[0m
[0;32mI (9306684) OPENTHREAD: Host update[0m
[0;32mI (9306684) OPENTHREAD: update host 5E3EECCCE3B4EBD1[0m
[0;32mI (9306684) OPENTHREAD: Add a new host, addr FD23:4D87:79DD:1:9F1F:6C0E:825F:2351[0m
[0;32mI (9306694) OPENTHREAD: Service 503C485E229E3C3F-8F387B87986EB146._matter._tcp[0m
[0;32mI (9306694) OPENTHREAD: Add new service[0m
[0;32mI (9306694) OPENTHREAD: Service subtype _I503C485E229E3C3F[0m
[0;32mI (9320724) OPENTHREAD: Host update[0m
[0;32mI (9320724) OPENTHREAD: update host 5E3EECCCE3B4EBD1[0m
[0;32mI (9320724) OPENTHREAD: Set addr of host to FD23:4D87:79DD:1:9F1F:6C0E:825F:2351[0m
[0;32mI (9320724) OPENTHREAD: Service D4FCF7A32090F51C._matterc._udp[0m
[0;32mI (9320724) OPENTHREAD: Add new service[0m
[0;31mE (9320724) mdns: mdns_service_add_for_host(5895): Cannot add more services[0m
[0;31mE (9320724) mdns: Cannot allocate memory (line: 5919, free heap: 167084 bytes)[0m
[0;31mE (9320724) OPENTHREAD: Failed to add service[0m
[0;33mW(9320734) OPENTHREAD:[W] SrpServer——: Send fail response: 2[0m
[0;32mI (9322544) OPENTHREAD: Host update[0m
[0;32mI (9322544) OPENTHREAD: update host 5E3EECCCE3B4EBD1[0m
[0;32mI (9322544) OPENTHREAD: Set addr of host to FD23:4D87:79DD:1:9F1F:6C0E:825F:2351[0m
[0;32mI (9322544) OPENTHREAD: Service D4FCF7A32090F51C._matterc._udp[0m
[0;32mI (9322544) OPENTHREAD: Add new service[0m
[0;31mE (9322544) mdns: mdns_service_add_for_host(5895): Cannot add more services[0m
[0;31mE (9322544) mdns: Cannot allocate memory (line: 5919, free heap: 167072 bytes)[0m
[0;31mE (9322554) OPENTHREAD: Failed to add service[0m
[0;33mW(9322554) OPENTHREAD:[W] SrpServer——: Send fail response: 2[0m
[0;32mI (9325674) OPENTHREAD: Host update[0m
[0;32mI (9325674) OPENTHREAD: update host 5E3EECCCE3B4EBD1[0m
[0;32mI (9325674) OPENTHREAD: Set addr of host to FD23:4D87:79DD:1:9F1F:6C0E:825F:2351[0m
[0;32mI (9325674) OPENTHREAD: Service D4FCF7A32090F51C._matterc._udp[0m
[0;32mI (9325674) OPENTHREAD: Add new service[0m
[0;31mE (9325674) mdns: mdns_service_add_for_host(5895): Cannot add more services[0m
[0;31mE (9325674) mdns: Cannot allocate memory (line: 5919, free heap: 167060 bytes)[0m
[0;31mE (9325674) OPENTHREAD: Failed to add service[0m
[0;33mW(9325674) OPENTHREAD:[W] SrpServer——: Send fail response: 2[0m
[... 83 lines cut ...]
### Steps to reproduce the behavior

1. pair few devices (it occures when 2 and more devices connected (sometimes it connects), cannot add more than 4)
2. initiate device pairing in HomeAssistant companion app
3. wait until pairing failes

### Project release version

ESP-THREAD-BR Release v1.1

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

15.1.1

### Shell

ZSH

zwx1995esp, 2025-01-02:
Hi, please enlarge the settings for the configuration: `MDNS_MAX_SERVICES`, and then have a try.

zwx1995esp, 2025-01-09:
Hi, we have already enlarged the default configuration for the max mdns service. Please see the commit: [4844efcf7aed4249dfd3585912c6ab8bb6b9a522](https://github.com/espressif/esp-thread-br/commit/4844efcf7aed4249dfd3585912c6ab8bb6b9a522). Will close this issue, but if you encounter any other problems, feel free to open another issue. Thanks a lot.

closed 2025-01-09 by zwx1995esp