espressif/esp-thread-br #117: W5500 version mismatched

orangeagain, 2025-01-04:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

connect eth by pass W5500

### Actual behavior (suspected bug)

W5500 version mismatched, expected 0x04, got 0
x0hx

### Error logs or terminal output

```shell
w5500.mac: W5500 version mismatched, expected 0x04, got 0
x0hx
E (526) w5500.mac: emac_w5500_init(826): verify chip ID failed     
I (526) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (526) esp_eth: esp_eth_driver_install(228): init mac failed      
ESP_ERROR_CHECK failed: esp_err_t 0x10a (ESP_ERR_INVALID_VERSION) at 0x4201b057
--- 0x4201b057: eth_start at D:/Espressif7/frameworks/esp-idf-v5.3.2/examples/common_components/protocol_examples_common/eth_connect.c:154 (discriminator 1)
 (inlined by) example_ethernet_connect at D:/Espressif7/frameworks/esp-idf-v5.3.2/examples/common_components/protocol_examples_common/eth_connect.c:232 (discriminator 1)

file: "/IDF/examples/common_components/protocol_examples_common/eth_connect.c" line 154
func: eth_start
expression: esp_eth_driver_install(&config, &s_eth_handle)

abort() was called at PC 0x4037c797 on core 0
.--- 0x4037c797: _esp_error_check_failed at D:/Espressif7/frameworks/esp-idf-v5.3.2/components/esp_system/esp_err.c:49

Backtrace: 0x40375b0a:0x3fcb6930 0x4037c7a1:0x3fcb6950 0x40383666:0x3fcb6970 0x4037c797:0x3fcb69e0 0x4201b057:0x3fcb6a10 0x4201aadb:0x3fcb6b30
.--- 0x40375b0a: panic_abort at D:/Espressif7/frameworks/esp-idf-v5.3.2/components/esp_system/panic.c:463
0x4037c7a1: esp_system_abort at D:/Espressif7/frameworks/esp-idf-v5.3.2/components/esp_system/port/esp_system_chip.c:92
0x40383666: abort at D:/Espressif7/frameworks/esp-idf-v5.3.2/components/newlib/abort.c:38
0x4037c797: _esp_error_check_failed at D:/Espressif7/frameworks/esp-idf-v5.3.2/components/esp_system/esp_err.c:49
0x4201b057: eth_start at D:/Espressif7/frameworks/esp-idf-v5.3.2/examples/common_components/protocol_examples_common/eth_connect.c:154 (discriminator 1)
 (inlined by) example_ethernet_connect at D:/Espressif7/frameworks/esp-idf-v5.3.2/examples/common_components/protocol_examples_common/eth_connect.c:232 (discriminator 1)
0x4201aadb: ot_br_init at D:/DanPianJi/test/esp-thread-br-main/examples/common/thread_border_router/src/border_router_launch.c:134
```

### Steps to reproduce the behavior

Enable EXAMPLE_CONNECT_ETHERNET
Disable EXAMPLE_CONNECT_WIFI

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

PowerShell

### Additional context

Use Defult hardware setting like this:
[image]

zwx1995esp, 2025-01-09:
Hi, @orangeagain From the logs you shared, I guess you used the tag `v5.3.2` for the IDF branch. And I tried the IDF tag `v5.3.2` with the latest esp-thread-br commit `1ce0e17792639dd50e145861f559009473f93ac7` no such issues on my side. Could you please clean your env and have a try again?

[image]

orangeagain, 2025-01-24:
解决了，漏掉了如下设置:
```
Espressif IoT Development Framework Configuration
[ ] connect using WiFi interface
[*] connect using Ethernet interface
(2048)  emac_rx task stack size
      Ethernet Type (W5500 Module)  --->
(2)     SPI Host Number
(21)    SPI SCLK GPIO number
(45)    SPI MOSI GPIO number
(38)    SPI MISO GPIO number
(41)    SPI CS GPIO number
(36)    SPI clock speed (MHz)
(39)    Interrupt GPIO number
(40)    PHY Reset GPIO number
(1)     PHY Address
[*] Obtain IPv6 address
     Preferred IPv6 Type (Local Link Address)  --->
```

closed 2025-01-24 by orangeagain