espressif/esp-thread-br #118: Inquiries about Coding for ESP Thread Border Router

Robbie981, 2025-01-08:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hi Espressif Team,

I'm working a project to build a thread based sensor network a have a few unanswered questions after going through the ESP Thread BR documentation:

1. My understanding is that the built RCP image will be automatically packed into the border router firmware. If I am creating my own rcp firmware (for the esp32-h2) how do I ensure that my new rcp firmware is being flashed? In other words, when I'm flashing the host SoC, how does it know which rcp firmware to flash?

2. This thread network demo video is really interesting (https://www.youtube.com/watch?v=0WXcu_r_lvQ&t=65s&ab_channel=EspressifSystems), however is there source code available for this? Or something similar perhaps if I am trying to recreate this?

Thank you!

Robbie981, 2025-01-09:
Hi, @zwx1995esp thank you for the response! A bit of an open-ended follow up question...

My understanding is that the 3 main sections of code relevant to the border router are in these directories: [esp-thread-br/examples/common/thread_border_router](https://github.com/espressif/esp-thread-br/tree/main/examples/common/thread_border_router), [esp-thread-br/examples/common/thread_border_router](https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router), and [esp-idf/examples/openthread/ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp). 

Practically speaking if I am working towards implementing a thread network where the border router is relaying CoAP/UDP messages between a host computer and thread end devices, where in the 3 directories I previous mentioned would I focus on for my intended implementation? I am asking this as I am not quite sure what the functional purposes of those directories are and how they work together.

It would be great if you could give me some guidance for this, thank you so much!

zwx1995esp, 2025-01-10:
Hi, @Robbie981  For relaying CoAP/UDP messages between a host computer and thread end devices, this feature has been supported at all. There might be nothing additionally you need to do for using this feature. We support IP layer forwarding, refer to [bi-directional-ipv6-connectivity](https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/connectivity.html#bi-directional-ipv6-connectivity). UDP and CoAP is upper layer based IP layer. You can have a try.

Here are some links for you which might be useful:
[Why can’t I access the host from the BR device using the ping command?](https://docs.espressif.com/projects/esp-thread-br/en/latest/qa.html#q1-why-can-t-i-access-the-host-from-the-br-device-using-the-ping-command)
UDP command: [`udpsockclient`](https://github.com/espressif/esp-thread-br/tree/main/components/esp_ot_cli_extension#udpsockclient) and [`udpsockserver`](https://github.com/espressif/esp-thread-br/tree/main/components/esp_ot_cli_extension#udpsockserver)
For the CoAP, we did not support any LwIP layer console command, you can refer to our [CoAP components](https://components.espressif.com/components/espressif/coap/versions/4.3.5~2) and try to implement it on OT device.

closed 2025-01-16 by Robbie981