espressif/esp-thread-br #119: WEB Gui - is there a way to password protect "active" actions and keep read-only mode for all other users?

pczekalski, 2025-01-15:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

I'm looking for a feature that at least lets me protect the WEB GUI from unauthorised users to modify the device's and network's configuration. It could be achieved either by full protection of the WEB GUI (at least with basic authentication) or, best, by providing two access layers: passive for browsing and active for developers, who might modify the device's and network's configuration. 

### Use cases

In laboratory scenarios, admins would like to provide regular users with open-thread network information as seen from the border router point of view (e.g., network topology) but not let them re-configure the network or the device via the WEB GUI. 

### Alternatives

Use of external authentication with e.g. nginx and reverse proxy, but that requires extra infrastructure.

zwx1995esp, 2025-01-21:
Hi @pczekalski your demand is something like an application layer feature. Our WEB server feature is designed based the compoment [`esp_http_server`](https://github.com/espressif/esp-idf/tree/master/components/esp_http_server). And for your question, you can create an issue in [IDF](https://github.com/espressif/esp-idf/issues) and ask someone who is responsable for http server.

zwx1995esp, 2025-03-25:
Consider to close this issue since there are no more comments. Feel free to re-open this if there are any updates.

closed 2025-03-25 by zwx1995esp