espressif/esp-thread-br #121: ESP32S3 is not able to load RCP image into ESP32H2

rajuga153, 2025-01-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi,

Today I have build the Thread broader router images and flashed into Dev Kit board. While booting I am seeing issue with load RCP image by UART0. Pls go through the log in below and let me know hoe to address this issue?

I (26) boot: ESP-IDF v5.5-dev-542-ge6aeb3d36d-dirty 2nd stage bootloader
I (26) boot: compile time Jan 16 2025 17:51:41
I (26) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (27) boot: efuse block revision: v1.3
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (27) boot.esp32s3: SPI Mode       : DIO
I (27) boot.esp32s3: SPI Flash Size : 4MB
I (28) boot: Enabling RNG early entropy source...
I (28) boot: Partition Table:
I (28) boot: ## Label            Usage          Type ST Offset   Length
I (28) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (29) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (29) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (30) boot:  3 ota_0            OTA app          00 10 00020000 00190000
I (30) boot:  4 ota_1            OTA app          00 11 001b0000 00190000
I (30) boot:  5 web_storage      Unknown data     01 82 00340000 00019000
I (31) boot:  6 rcp_fw           Unknown data     01 82 00359000 000a0000
I (31) boot: End of partition table
I (31) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=469ach (289196) map
I (83) esp_image: segment 1: paddr=000669d4 vaddr=3fc99900 size=04efch ( 20220) load
I (87) esp_image: segment 2: paddr=0006b8d8 vaddr=40374000 size=04740h ( 18240) load
I (92) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=f50c0h (1003712) map
I (269) esp_image: segment 4: paddr=001650e8 vaddr=40378740 size=11100h ( 69888) load
I (284) esp_image: segment 5: paddr=001761f0 vaddr=600fe100 size=0001ch (    28) load
I (293) boot: Loaded app from partition at offset 0x20000
I (293) boot: Disabling RNG early entropy source...
I (294) cpu_start: Multicore app
I (303) cpu_start: Pro cpu start user code
I (303) cpu_start: cpu freq: 160000000 Hz
I (303) app_init: Application information:
I (304) app_init: Project name:     esp_ot_br
I (304) app_init: App version:      v5.4-beta2-16-g4844efc
I (304) app_init: Compile time:     Jan 16 2025 17:49:38
I (304) app_init: ELF file SHA256:  94f9e12c0...
I (304) app_init: ESP-IDF:          v5.5-dev-542-ge6aeb3d36d-dirty
I (305) efuse_init: Min chip rev:     v0.0
I (305) efuse_init: Max chip rev:     v0.99
I (305) efuse_init: Chip rev:         v0.2
I (305) heap_init: Initializing. RAM available for dynamic allocation:
I (305) heap_init: At 3FCAD870 len 0003BEA0 (239 KiB): RAM
I (305) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (306) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (306) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (307) spi_flash: detected chip: gd
I (307) spi_flash: flash io: dio
W (307) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (308) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (309) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (310) main_task: Started on CPU0
I (320) main_task: Calling app_main()
I (370) RCP_UPDATE: RCP: using update sequence 1
I (370) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (370) OPENTHREAD: spinel UART interface initialization completed
I (370) main_task: Returned from app_main()
W(2380) OPENTHREAD:[W] P-SpinelDrive-: Wait for response timeout
W (2380) uart: GPIO 18 is not usable, maybe used by others
I (2380) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (2380) esp_ot_br: Internal RCP Version: openthread-esp32/e6aeb3d36d-005c5cefc; esp32h2;  2025-01-16 12:03:42 UTC
W (2380) uart: GPIO 18 is not usable, maybe used by others
I (2380) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (2390) gpio: GPIO[7]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2390) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (2760) RCP_UPDATE: Erasing flash (this may take a while)...
DEBUG: Flash size detection failed, falling back to default
I (2940) RCP_UPDATE: Start programming
I (2940) RCP_UPDATE: binary_size 19488
I (2970) RCP_UPDATE: left size 18464, written 1024
I (2970) RCP_UPDATE: Progress: 5 %
I (3000) RCP_UPDATE: left size 17440, written 2048
I (3000) RCP_UPDATE: Progress: 10 %
I (3030) RCP_UPDATE: left size 16416, written 3072
I (3040) RCP_UPDATE: Progress: 15 %
I (3070) RCP_UPDATE: left size 15392, written 4096
I (3070) RCP_UPDATE: Progress: 21 %
I (3100) RCP_UPDATE: left size 14368, written 5120
I (3100) RCP_UPDATE: Progress: 26 %
I (3130) RCP_UPDATE: left size 13344, written 6144
I (3140) RCP_UPDATE: Progress: 31 %
I (3170) RCP_UPDATE: left size 12320, written 7168
I (3170) RCP_UPDATE: Progress: 36 %
I (3200) RCP_UPDATE: left size 11296, written 8192
I (3200) RCP_UPDATE: Progress: 42 %
I (3240) RCP_UPDATE: left size 10272, written 9216
I (3240) RCP_UPDATE: Progress: 47 %
I (3270) RCP_UPDATE: left size 9248, written 10240
I (3270) RCP_UPDATE: Progress: 52 %
I (3300) RCP_UPDATE: left size 8224, written 11264
I (3300) RCP_UPDATE: Progress: 57 %
I (3340) RCP_UPDATE: left size 7200, written 12288
I (3340) RCP_UPDATE: Progress: 63 %
I (3370) RCP_UPDATE: left size 6176, written 13312
I (3370) RCP_UPDATE: Progress: 68 %
I (3400) RCP_UPDATE: left size 5152, written 14336
I (3400) RCP_UPDATE: Progress: 73 %
I (3440) RCP_UPDATE: left size 4128, written 15360
I (3440) RCP_UPDATE: Progress: 78 %
I (3470) RCP_UPDATE: left size 3104, written 16384
I (3470) RCP_UPDATE: Progress: 84 %
I (3500) RCP_UPDATE: left size 2080, written 17408
I (3500) RCP_UPDATE: Progress: 89 %
I (3540) RCP_UPDATE: left size 1056, written 18432
I (3540) RCP_UPDATE: Progress: 94 %
I (3570) RCP_UPDATE: left size 32, written 19456
I (3570) RCP_UPDATE: Progress: 99 %
I (3600) RCP_UPDATE: left size 0, written 19488
I (3600) RCP_UPDATE: Progress: 100 %
I (3600) RCP_UPDATE: Finished programming
I (3650) RCP_UPDATE: Flash verified
I (3650) RCP_UPDATE: Erasing flash (this may take a while)...
DEBUG: Flash size detection failed, falling back to default
I (4630) RCP_UPDATE: Start programming
I (4640) RCP_UPDATE: binary_size 201712
I (4670) RCP_UPDATE: left size 200688, written 1024
I (4670) RCP_UPDATE: Progress: 0 %
I (4700) RCP_UPDATE: left size 199664, written 2048
I (4700) RCP_UPDATE: Progress: 1 %
I (4730) RCP_UPDATE: left size 198640, written 3072
I (4730) RCP_UPDATE: Progress: 1 %
I (4770) RCP_UPDATE: left size 197616, written 4096
I (4770) RCP_UPDATE: Progress: 2 %
I (4800) RCP_UPDATE: left size 196592, written 5120

zwx1995esp, 2025-01-21:
Hi, @rajuga153 These logs should be normal for the RCP update processing. So, after rcp updating, can the br work well?

rajuga153, 2025-01-21:
Hi, 

Above log is repeating continuous and br functionality is not started.

zwx1995esp, 2025-01-21:
Hi, @rajuga153  that is too odd. Could you please share the code base you used? Which IDF and esp-thread-br branch & commit hash you used? Did you use our [ESP Thread Border Router Board](https://github.com/espressif/esp-thread-br?tab=readme-ov-file#esp-thread-border-router-board) to run this example?

rajuga153, 2025-01-28:
Hi @zwx1995esp ,

Sorry for the late reply. Pls find required details in below.

ESP-IDF: git checkout e6aeb3d36d2fcaf578e4e53308191a8e262dc822
ESP-Thread-BR: git clone --recursive https://github.com/espressif/esp-thread-br.git

[image]

chshu, 2025-02-08:
@rajuga153 Could you share the full log and the built binary? The [basic_thread_border_router](https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router) exampe should work as is on that board.

zwx1995esp, 2025-03-06:
Hi, I will close this issue since there are no further comments. Feel free to reopen it if you encounter any related issues.

closed 2025-03-06 by zwx1995esp