espressif/esp-thread-br #123: How to get cli input or output over mqtt

orangeagain, 2025-01-21:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

I need get information via mqtt, so I want to send command to cli component, and get output text

### Use cases

mqtt remote debug

zwx1995esp, 2025-01-21:
Hi @orangeagain Let me try to understand what you want to do. For example the command `thread start`:

Do you mean you want to send a message using mqtt to otbr and otbr can process the same function just like inputting the command `thread start` using ot cli?

If what I understand is right, you can refer to `https://openthread.io/reference` For example [`thread start`](https://openthread.io/reference/cli/commands#thread_start)

[image]

You can search the related API: otThreadSetEnabled in openthread code.

Notice: [esp_openthread_lock_acquire](https://github.com/espressif/esp-idf/blob/master/components/openthread/include/esp_openthread_lock.h#L48)

zwx1995esp, 2025-01-21:
Or if you want to receive some command from mqtt message and run these commands on otbr, you can refer to our [cli task](https://github.com/espressif/esp-idf/blob/master/components/openthread/src/esp_openthread_cli.c) source code.

orangeagain, 2025-01-22:
> Or if you want to receive some command from mqtt message and run these commands on otbr, you can refer to our [cli task](https://github.com/espressif/esp-idf/blob/master/components/openthread/src/esp_openthread_cli.c) source code.

cool , this is what i need

closed 2025-01-22 by orangeagain