espressif/esp-thread-br #124: 在普通 ESP32S3开发板上无法运行

orangeagain, 2025-01-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

期待控制台在启动后有输出

### Actual behavior (suspected bug)

只打印 bootloader第一阶段的数据，然后卡住，然后看门狗超时不断重启。

### Error logs or terminal output

```shell
···
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x1870
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc40
load:0x403cb700,len:0x2ee8
entry 0x403c8908
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759a9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x1870
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc40
load:0x403cb700,len:0x2ee8
entry 0x403c8908
···
```

### Steps to reproduce the behavior

1.设定最新提交项目basic_ot_br为esp32s3
2.连接esp32s3 n16r8开发板 或 自制esp32s3N16R8板子
3.启动卡住。在自制板子上只打印第一阶段bootloader文本，在开发板上则无输出。

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Win 10

### Shell

PowerShell

orangeagain, 2025-01-23:
> 请问，您使用的是S3开发板/自制的开发板的UART端还是USB端？

UART端，CH340C串口芯片。
换别的项目可以烧录有输出，就这个项目可以烧录但没输出，说明硬件应该没问题
看打印信息，似乎没有进入第二启动阶段(SBL)

zwx1995esp, 2025-01-24:
> UART端，CH340C串口芯片。 换别的项目可以烧录有输出，就这个项目可以烧录但没输出，说明硬件应该没问题 看打印信息，似乎没有进入第二启动阶段(SBL)

esp-thread-br 默认使用的是USB，如果需要用 UART ，请配置

Component config → ESP System Settings → Channel for console output → Default: UART0

[image]

closed 2025-01-24 by orangeagain

zwx1995esp, 2025-01-26:
> 确实如你所说。问题解决了。令人疑惑的是br板的原理图也是uart0引脚连接串口芯片的，为什么那个就可以不改运行

Hi @orangeagain BR 开发板上是USB接口，并非UART接口，所以不改就可以运行。