espressif/esp-thread-br #125: Esp32-s3, H-2 combo board unable to communicate with eachother

benjaminhope, 2025-01-23:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The s3 to communicate with the h2 and continue with booting.

### Actual behavior (suspected bug)

Communication error between the two chips and abort into reboot.

### Error logs or terminal output

```shell
W(2543) OPENTHREAD:[W] P-SpinelDrive-: Wait for response timeout
W(4553) OPENTHREAD:[W] P-SpinelDrive-: Wait for response timeout
E(4553) OPENTHREAD:[C] P-SpinelDrive-: Failed to reset co-processor!
E(4553) OPENTHREAD:[C] Platform------: ResetCoprocessor() at spinel_driver.cpp:160: Failure
```

### Steps to reproduce the behavior

Install ESP-IDF v5.4 with visual studio code.
Run ot_br and ot_rcp example
Build and flash

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

24H2

### Shell

other (details in Additional context)

### Additional context

Visual Studio ESP-IDF extension

chshu, 2025-02-08:
@benjaminhope The issue typically occurs when the UART configuration does not match the actual connected pins. You may need to change the default uart pin numbers here: https://github.com/espressif/esp-idf/blob/master/examples/openthread/ot_br/main/esp_ot_config.h#L40

zwx1995esp, 2025-03-06:
Hi, I will close this issue since no more comments. But feel free to reopen this if you encouneter any issues related.

closed 2025-03-06 by zwx1995esp