espressif/esp-thread-br #127: ESP Border Router HTTP Requests Timing Out

Robbie981, 2025-02-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

After the border router connects to the WiFi (I can confirm this is connected as I am able to access the web GUI), I have code that tries to send an http POST request to `https://httpbin.org/` for testing purposes, however I keep getting a timeout error `W (53653) HTTP_CLIENT: Connection timed out before data was ready!` What could be the issue here?

I've tried following a similar approach in my code following [this example code](https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_http_client/main/esp_http_client_example.c). My code for the http request is posted below.

[image]

gytxxsy, 2025-03-07:
Hi @Robbie981 . This issue appears to be an application-layer problem related to the [esp_http_client](https://github.com/espressif/esp-idf/tree/master/components/esp_http_client). You can create a new issue in [IDF](https://github.com/espressif/esp-idf/issues) and ask someone who is responsable for http server. Thanks.

zwx1995esp, 2025-03-25:
Hi, closes this issue since no more comments. Feel free to re-open if you encountered any issues related or have any updates. Thanks.

closed 2025-03-25 by zwx1995esp