espressif/esp-thread-br #128: ssue with Setting Leader Weight and Router Downgrade Threshold in OpenThread (IDFGH-14734)

donaldkohcw, 2025-02-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 

I am currently working on an OpenThread using ESP-IDF v5.3.2 and I am trying to set the leader weight and router downgrade threshold in my OpenThread network. However, I am encountering some issues with the otOperationalDataset structure. Is following function supported in esp ot api?

    // Set the leader weight to 255
    otThreadSetLocalLeaderWeight(esp_openthread_get_instance(), 255);

    // Set the RouterDowngradeThreshold to 0
    otThreadSetRouterDowngradeThreshold(esp_openthread_get_instance(), 0);

I got following error:

C:/Espressif/frameworks/esp-idf-v5.3.2/examples/openthread/ot_cli_lead/main/esp_ot_cli.c:106:24: error: 'otOperationalDatasetComponents' has no member named 'mIsLeaderWeightPresent'
  106 |     dataset.mComponents.mIsLeaderWeightPresent = true;
      |                        ^
C:/Espressif/frameworks/esp-idf-v5.3.2/examples/openthread/ot_cli_lead/main/esp_ot_cli.c:107:12: error: 'otOperationalDataset' has no member named 'mLeaderWeight'
  107 |     dataset.mLeaderWeight = 255;
      |            ^
C:/Espressif/frameworks/esp-idf-v5.3.2/examples/openthread/ot_cli_lead/main/esp_ot_cli.c:110:24: error: 'otOperationalDatasetComponents' has no member named 'mIsRouterDowngradeThresholdPresent'
  110 |     dataset.mComponents.mIsRouterDowngradeThresholdPresent = true;
      |                        ^
C:/Espressif/frameworks/esp-idf-v5.3.2/examples/openthread/ot_cli_lead/main/esp_ot_cli.c:111:12: error: 'otOperationalDataset' has no member named 'mRouterDowngradeThreshold'
  111 |     dataset.mRouterDowngradeThreshold = 0;

Thanks,

zwx1995esp, 2025-02-27:
Hi, @donaldkohcw I have a question: why you need to call the function `otCliInputLine` directly? This function is called by [ot_cli_task ](https://github.com/espressif/esp-idf/blob/master/components/openthread/src/esp_openthread_cli.c#L49)to handle command input by the console.

If you really need to call this function in some other tasks, please acquire the [ot lock](https://github.com/espressif/esp-idf/blob/master/components/openthread/include/esp_openthread_lock.h#L48) first.

zwx1995esp, 2025-02-27:
Hi, @donaldkohcw  The recommanded way is that calling the Openthread APIs directly. As for the previous issue you shared:
> C:/Espressif/frameworks/esp-idf-v5.3.2/examples/openthread/ot_cli_lead/main/esp_ot_cli.c:106:24: error: 'otOperationalDatasetComponents' has no member named 'mIsLeaderWeightPresent'
106 | dataset.mComponents.mIsLeaderWeightPresent = true;

So, did you add these codes in your own application? I did not see the member variable `mIsLeaderWeightPresent ` for the struct `[otOpeartionDataset](https://github.com/openthread/openthread/blob/main/include/openthread/dataset.h#L221)`. Could you please check? If these codes are added by accident, please remove them.
```
106 | dataset.mComponents.mIsLeaderWeightPresent = true;
107 | dataset.mLeaderWeight = 255;
110 | dataset.mComponents.mIsRouterDowngradeThresholdPresent = true;
111 | dataset.mRouterDowngradeThreshold = 0;
```

These two APIs can be called to set the related configurations you need, (Notice: if you call these functions out of thread task context, please also acquire the ot lock mentioned in the previous comment):
```
// Set the leader weight to 255
otThreadSetLocalLeaderWeight(esp_openthread_get_instance(), 255);

// Set the RouterDowngradeThreshold to 0
otThreadSetRouterDowngradeThreshold(esp_openthread_get_instance(), 0);
```

dk77148, 2025-02-28:
Hi @zwx1995esp,

Thanks for spotting it and for the guide! 👍 I got this running.

Regards,
Don

zwx1995esp, 2025-03-03:
Hi, @dk77148  Great! Since your issue has beed fixed, I will close this Issue. If you encounter any issues related, feel free to re-open it.

closed 2025-03-03 by zwx1995esp