espressif/esp-thread-br #131: esp-thread-br\examples\basic_thread_border_router ：   Validate the IPv6 Connectivity fail

hsw-li, 2025-03-03:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

only once

### Expected behavior

 ipv6能够ping。或者能够提供一个ipv6验证可行的版本

### Actual behavior (suspected bug)

1.目前使用windows11,安装vscode，ESP-IDF的插件，下载的是master(v5.5.0),开发板是官方的
2.开发板是ESP Thread Border Router/Zigbee Gateway v1.2
3.目前没做任何修改，只跑了一遍例程。但是编译烧录，配置wifi后，主机端执行 ping fd1a:7ade:6191:73ea:0:ff:fe00:fc11 失败

### Error logs or terminal output

```shell
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037b16a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x15b4
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc80
load:0x403cb700,len:0x2f30
entry 0x403c8914
I (26) boot: ESP-IDF v5.5-dev-542-ge6aeb3d36d-dirty 2nd stage bootloader
I (27) boot: compile time Feb 28 2025 19:37:09
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (27) boot: efuse block revision: v1.3
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (28) boot.esp32s3: SPI Mode       : DIO
I (28) boot.esp32s3: SPI Flash Size : 4MB
I (28) boot: Enabling RNG early entropy source...
I (28) boot: Partition Table:
I (28) boot: ## Label            Usage          Type ST Offset   Length
I (29) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (29) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (30) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (30) boot:  3 ota_0            OTA app          00 10 00020000 00190000
I (31) boot:  4 ota_1            OTA app          00 11 001b0000 00190000
I (31) boot:  5 web_storage      Unknown data     01 82 00340000 00019000
I (31) boot:  6 rcp_fw           Unknown data     01 82 00359000 000a0000
I (32) boot: End of partition table
I (32) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=4681ch (288796) map
I (83) esp_image: segment 1: paddr=00066844 vaddr=3fc99900 size=04efch ( 20220) load
I (88) esp_image: segment 2: paddr=0006b748 vaddr=40374000 size=048d0h ( 18640) load
I (92) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=f4cb4h (1002676) map
I (269) esp_image: segment 4: paddr=00164cdc vaddr=403788d0 size=10f70h ( 69488) load
I (285) esp_image: segment 5: paddr=00175c54 vaddr=600fe100 size=0001ch (    28) load
I (294) boot: Loaded app from partition at offset 0x20000
I (294) boot: Disabling RNG early entropy source...
I (295) cpu_start: Multicore app
I (304) cpu_start: Pro cpu start user code
I (304) cpu_start: cpu freq: 160000000 Hz
I (304) app_init: Application information:
I (304) app_init: Project name:     esp_ot_br
I (304) app_init: App version:      v5.4-beta2-24-g8d163b7
I (305) app_init: Compile time:     Feb 28 2025 19:36:36
I (305) app_init: ELF file SHA256:  8d8838fee...
--- Warning: Checksum mismatch between flashed and built applications. Checksum of built application is 3cf758f111dd9d40dd27878fff05f332ead43608a0bfaa1e48a4891876bcac98
[... 148 lines cut ...]

### Project release version

v5.5.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

win11

### Shell

PowerShell

### Additional context

所有的操作，都按照官方的操作去实践的https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html
另外，有用linux主机尝试过，还是无法ping

zwx1995esp, 2025-03-03:
Hi, 您好，如您想验证双向IPv6链接功能，请参考我们官方文档的[codelab: connectivity](https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/connectivity.html) ，1.其中有一些配置需要设置在主机上。 2.双向IPv6的链接功能需要有一个Thread end Device一起完成测试和验证。

Tips: 1.根据您所描述的log，这个IPv6前缀: fd01:24ec:dfd9:355b::/64 是Thread网络的local前缀，windows的host无法访问这个前缀下的地址。可以尝试ping fde4:28d:9692:1:2baf:237:a2ea:4c7a. 这个是一个全局可访问的地址。
2. 请参考: [为什么BR和主机之前不能ping通](https://docs.espressif.com/projects/esp-thread-br/en/latest/qa.html#q1-why-can-t-i-access-the-host-from-the-br-device-using-the-ping-command)

hsw-li, 2025-03-03:
> Hi, [@hsw-li](https://github.com/hsw-li) 您好，如您想验证双向IPv6链接功能，请参考我们官方文档的[codelab: connectivity](https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/connectivity.html) ，1.其中有一些配置需要设置在主机上。 2.双向IPv6的链接功能需要有一个Thread end Device一起完成测试和验证。
> 
> Tips: 1.根据您所描述的log，这个IPv6前缀: fd01:24ec:dfd9:355b::/64 是Thread网络的local前缀，windows的host无法访问这个前缀下的地址。可以尝试ping fde4:28d:9692:1:2baf:237:a2ea:4c7a. 这个是一个全局可访问的地址。 2. 请参考: [为什么BR和主机之前不能ping通](https://docs.espressif.com/projects/esp-thread-br/en/latest/qa.html#q1-why-can-t-i-access-the-host-from-the-br-device-using-the-ping-command)

你好，我听了你的建议，用了新的地址  fde4:28d:9692:1:2baf:237:a2ea:4c7a，可以从主机ping到br设备，但还是无法解决我们的问题。（我重启了设备，地址有了一些变化）
1.我们需要通过matter来连接br，需要 dataset active -x  收集到的数据集。这个似乎不是fde4:*****地址有关的。

> ipaddr

fdc5:8ae7:8cc5:fbf3:0:ff:fe00:fc11
fde4:28d:9692:1:2baf:237:a2ea:4c7a
fdc5:8ae7:8cc5:fbf3:0:ff:fe00:fc10
fdc5:8ae7:8cc5:fbf3:0:ff:fe00:fc38
fdc5:8ae7:8cc5:fbf3:0:ff:fe00:fc00
fdc5:8ae7:8cc5:fbf3:0:ff:fe00:a800
fdc5:8ae7:8cc5:fbf3:843b:1:d00f:7db8
fe80:0:0:0:1c18:488a:c777:3786
Done
> dataset active -x

0e08000000000001000000030000184a0300000f35060004001fffe00208ddcd68864227b37d0708fdc58ae78cc5fbf30510831afe370bf84a1608c6c7a161cb6f61030f4f70656e5468726561642d633732300102c7200410ef5f3a75e22dfd2799365931c725520a0c0402a0f7f8
Done
> dataset active

Active Timestamp: 1
Channel: 24
Wake-up Channel: 15
Channel Mask: 0x07fff800
Ext PAN ID: ddcd68864227b37d
Mesh Local Prefix: fdc5:8ae7:8cc5:fbf3::/64
Network Key: 831afe370bf84a1608c6c7a161cb6f61
Network Name: OpenThread-c720
PAN ID: 0xc720
PSKc: ef5f3a75e22dfd2799365931c725520a
Security Policy: 672 onrc 0
Done

2.我真正的需要是需要如何通过br设备的ipv6地址并生成对应的数据集，让我能通过数据集实现matter和br设备的连接

zwx1995esp, 2025-03-03:
您好，如您所述:
> 我们需要通过matter来连接br，需要 dataset active -x 收集到的数据集。
> 我真正的需要是需要如何通过br设备的ipv6地址并生成对应的数据集，让我能通过数据集实现matter和br设备的连接

1.请问一下您这边的测试拓扑是什么样的？是有一个matter设备需要加入到BR的thread网络中吗？如果您能够提供一下具体的应用场景，比如有几台设备，每台设备之间用什么网络层链接，以及需要主机访问哪一个设备。我们可以更好的讨论一下细节。

2.dataset active -x这个返回的是整个thread网络信息，这个网络信息和设备上的地址没有直接的关系。dataset active -x这个返回的dataset可以用于其他设备加入到当前设备所创建的Thread网络里。

zwx1995esp, 2025-03-04:
> 1.目前只有有一台matter主机，然后和br设备在同一网段中。目前我这百年chip-tool测试成功，之前可能是同事的sdk版本有问题。也可能是chip-tool的问题，因为我是直接ubuntu snap安装的，同事是自己编译的。

Ok,所以说现在是没问题了嘛？

> 2.还有一个问题请教，我这边目前主机去ping fde4:******是可以成功的，但是从br设备去ping ubuntu主机，显示是失败，我是按照了文档中做了accept_ra 和accept_ra_rt_info_max_plen 配置的。想请教一下，是我的ipv6地址有问题吗？

这个可以参考之前我贴的链接

> 2. 请参考: [为什么BR和主机之前不能ping通](https://docs.espressif.com/projects/esp-thread-br/en/latest/qa.html#q1-why-can-t-i-access-the-host-from-the-br-device-using-the-ping-command)

> 3.图中红色1的位置显示是ipaddr的第一个地址， 对于一些ipv6不熟悉的人，是不是会产生误导

感谢您对文档提出的修改意见，我们会添加一些详细的关于ipv6的说明。可以在thread end device上使用如下指令:
```
ipaddr -v
```
显示更多有关地址的信息。
```
fd71:669c:bb6:1:530f:9ad1:b62e:3eda **origin:slaac** plen:64 preferred:1 valid:1
fdde:ad00:beef:0:0:ff:fe00:fc00 origin:thread plen:64 preferred:1 valid:1
fdde:ad00:beef:0:0:ff:fe00:a400 origin:thread plen:64 preferred:1 valid:1
fdde:ad00:beef:0:3082:6585:d40d:5197 origin:thread plen:64 preferred:1 valid:1
fe80:0:0:0:2c4e:c2b:aa13:bce2 origin:thread plen:64 preferred:1 valid:1
```
使用带有 origin:slaac 这个属性的地址来ping.    origin:thread 属性的地址是thread内部的地址，只能和thread网络内的设备通信。slaac是静态添加的global unicast 地址。

hsw-li, 2025-03-04:
> Ok,所以说现在是没问题了嘛？
>
>
> 这个可以参考之前我贴的链接
>
>
>
> 感谢您对文档提出的修改意见，我们会添加一些详细的关于ipv6的说明。可以在thread end device上使用如下指令:
>
> ```
> ipaddr -v
> ```
>
> 显示更多有关地址的信息。
>
> ```
> fd71:669c:bb6:1:530f:9ad1:b62e:3eda **origin:slaac** plen:64 [...]

1.谢谢，目前是没问题了。
2.关于br为什么不能ping 主机，我刚又看了认真FAQ，确实解答了我的问题。不好意思，昨天这部分没有认真看。
3.谢谢你对ipv6相关知识的指导，后续我会研究一下这些属性标签的区别

另外关于LWIP层面实现ping的问题，后续官方会进行添加吗？还是说交给用户按自己需求实现？

你回答后，我会关掉这个问题

zwx1995esp, 2025-03-04:
> 另外关于LWIP层面实现ping的问题，后续官方会进行添加吗？还是说交给用户按自己需求实现？

Hi, ICMPv6 Ping 的收发有一些自定义的数据和重传等机制可以自定义实现，这个交给用户按自己需求实现，我们目前没有计划支持。

由于目前问题已经解决了，关掉这个issue，如果您还有其他相关问题，可以重新打开这个issue。若有新的问题，也可以新开issue。谢谢。

closed 2025-03-04 by zwx1995esp

hsw-li, 2025-03-05:
hi，今天我们重新测试的过程中，有一些新的疑问，想请教一下。
我这边测试过程中，发现在主机的wifi接入2.4G的情况下，是可以ping通的，但是如果主机接入的是5G的频段，我们去ping同一个地址就会失败，当然也就无法进行设备的接入。所以想请教一下，这和wifi的频段有关系吗？

zwx1995esp, 2025-03-06:
> hi，今天我们重新测试的过程中，有一些新的疑问，想请教一下。 我这边测试过程中，发现在主机的wifi接入2.4G的情况下，是可以ping通的，但是如果主机接入的是5G的频段，我们去ping同一个地址就会失败，当然也就无法进行设备的接入。所以想请教一下，这和wifi的频段有关系吗？

Hi, 您好，S3不支持 5G 频段。所以连不上 5G 的wifi。

hsw-li, 2025-03-06:
> > hi，今天我们重新测试的过程中，有一些新的疑问，想请教一下。 我这边测试过程中，发现在主机的wifi接入2.4G的情况下，是可以ping通的，但是如果主机接入的是5G的频段，我们去ping同一个地址就会失败，当然也就无法进行设备的接入。所以想请教一下，这和wifi的频段有关系吗？
> 
> Hi, 您好，S3不支持 5G 频段。所以连不上 5G 的wifi。

1.好的，但我现在是br设备接入路由器2.4G, 主机接入同一个ssid的5G频段。
br---链接--- a_2.4G        主机----链接---a_5G，中间的数据交换不应该由路由器转发吗？
流程不应该是主机---->路由----->br设备，路由会主动转换到对应的频段才对。
从流程上看，S3不支持5G频段,和我们要做的事并没有直接关系

2.另外有个问题想请教，和这个主题无关，本来是向在esp论坛请教，但是注册没办法通过。
我注意到issue中有些人也发送了相关的问题，但是好像都没有达到我的要求。又或者我的英文对与某些问题下的回答实在是没看懂，所以就直接请教了
目前br设备启动，log和console会从usb一起输出，进行交互。但是我想让console和log分开，尤其是初始化的信息，我想输出到另一个uart,然后usb单独做console,usb这边除了我下达command会输出的信息，启动时刷新的信息都不要输出。（目前我借助menuconfig中consle 二次辅助output接口，可以做到usb输出信息不进行console，uart输出信息和console交互,但是uart还是会输出信息）
示意图如下，我想实现第二种效果

[image]

如果曾经又人提问过，又有对应的方法，请贴个链接就好。

zwx1995esp, 2025-03-06:
> 1.好的，但我现在是br设备接入路由器2.4G, 主机接入同一个ssid的5G频段。 br---链接--- a_2.4G 主机----链接---a_5G，中间的数据交换不应该由路由器转发吗？ 流程不应该是主机---->路由----->br设备，路由会主动转换到对应的频段才对。 从流程上看，S3不支持5G频段,和我们要做的事并没有直接关系

这个要看路由器有没有转发，如果路由器有转发我们的设备就可以完成如您所说的通信。

> 2.另外有个问题想请教，和这个主题无关，本来是向在esp论坛请教，但是注册没办法通过。 我注意到issue中有些人也发送了相关的问题，但是好像都没有达到我的要求。又或者我的英文对与某些问题下的回答实在是没看懂，所以就直接请教了 [...]

抱歉，这个我不是很了解，可以去ESP-IDF这个repo上开个issue找相关负责人问一下。

hsw-li, 2025-03-06:
https://github.com/espressif/esp-thread-br/issues/131#issuecomment-2702983296

好的，多谢