espressif/esp-thread-br #132: udp/coap receiver issue in linux with esp ot_rcp

donaldkohcw, 2025-03-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi,

I am facing an issue with the Linux receiver: the Linux terminal is not receiving any inbound messages via UDP/CoAP, even though it has no problem sending out messages via UDP/CoAP to other children/routers.

I can ping between devices without any issues, which means that both directions are reachable.

I tried running "sudo tcpdump -ni wpan0 udp port 12345" but seems like it never reaches. If i just run on another terminal in the same linux locally, "echo "Hello, UDP!" | nc -u 127.0.0.1 12345" , it received. 

It is set up with ot_rcp idf 5.32v and Linux Ubuntu.

I have also tried disabling the firewall using "sudo ufw disable." Do you have any tips for resolving this? Thanks.

Regards,
dk

donaldkohcw, 2025-03-04:
I checked the firewall on linux, seems like everything is opened.

sudo iptables -L -n -v
Chain INPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 ACCEPT     0    --  *      ens33   0.0.0.0/0            0.0.0.0/0           
    0     0 ACCEPT     0    --  ens33  *       0.0.0.0/0            0.0.0.0/0           

Chain OUTPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination

donaldkohcw, 2025-03-04:
I tried to listen on all interface if there is any packet but failed to capture any.

sudo tcpdump -i any udp port 12345 -v
tcpdump: data link type LINUX_SLL2
tcpdump: listening on any, link-type LINUX_SLL2 (Linux cooked v2), snapshot length 262144 bytes

donaldkohcw, 2025-03-04:
below is my linux terminal ipaddr. I tried on all of them but failed as well. 

From esp32h2 installed with ot-cli, I udp send fdc5:61d5:bd6e:1:c441:9814:bcaa:e476 12345 "hello"

donaldkoh@donaldkoh-VMware-Virtual-Platform:~$ sudo ot-ctl ipaddr
fdda:a4dd:cf3a:f260:0:ff:fe00:fc11
fdc5:61d5:bd6e:1:c441:9814:bcaa:e476
fdda:a4dd:cf3a:f260:0:ff:fe00:fc10
fdda:a4dd:cf3a:f260:0:ff:fe00:fc38
fdda:a4dd:cf3a:f260:0:ff:fe00:fc00
fdda:a4dd:cf3a:f260:0:ff:fe00:6c00
fdda:a4dd:cf3a:f260:97b6:23d9:8862:5103
fe80:0:0:0:4490:a449:353d:3743
Done

chshu, 2025-03-06:
@donaldkohcw `ot_rcp` handles the 802.15.4 mac layer packets only, if the ping works, it should not drop the UDP/CoAP packets.

Pleasse double check the following:
-  the `ot-cli` joins the same network of the BR (Linux Host + ot_rcp), and confirm the two can ping each other. 
- run `ifconfig` on Linux, and confirm the ipv6 address is listed in the wpan0 interface

Then send UDP using the [udpsockclient](https://github.com/espressif/esp-thread-br/tree/main/components/esp_ot_cli_extension#udpsockclient) commands from esp ot-cli:

```
udpsockclient open
udpsockclient send fdc5:61d5:bd6e:1:c441:9814:bcaa:e476 12345 hell
```

The tcpdump should capture the UDP frame.

dk77148, 2025-03-06:
Hi @chshu, 

Thanks for following up. I am able to see udp message (after i udp send)during monitoring obtr activites "sudo journalctl -u otbr-agent.service -f"

Mar 06 10:20:51 donaldkoh-VMware-Virtual-Platform otbr-agent[3554]: 01:29:20.792 [I] MeshForwarder-: src:[fd3a:231d:458b:9544:8d64:a07a:93d2:6ae]:49155
Mar 06 10:20:51 donaldkoh-VMware-Virtual-Platform otbr-agent[3554]: 01:29:20.793 [I] MeshForwarder-: dst:[fd3a:231d:458b:9544:97b6:23d9:8862:5103]:12345

I did ifconfig and all the ip address are tied to wpan0

donaldkoh@donaldkoh-VMware-Virtual-Platform:~/ot-br-posix$ ifconfig wpan0
wpan0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1280
        inet6 fd3a:231d:458b:9544:0:ff:fe00:fc11  prefixlen 64  scopeid 0x0<global>
        inet6 fd3a:231d:458b:9544:0:ff:fe00:6c00  prefixlen 64  scopeid 0x0<global>
        inet6 fd3a:231d:458b:9544:0:ff:fe00:fc10  prefixlen 64  scopeid 0x0<global>
        inet6 fd3a:231d:458b:9544:0:ff:fe00:fc38  prefixlen 64  scopeid 0x0<global>
        inet6 fd3a:231d:458b:9544:97b6:23d9:8862:5103  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::4490:a449:353d:3743  prefixlen 64  scopeid 0x20<link>
        inet6 fdc5:61d5:bd6e:1:c441:9814:bcaa:e476  prefixlen 64  scopeid 0x0<global>
        inet6 fd3a:231d:458b:9544:0:ff:fe00:fc00  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 1552  bytes 139863 (139.8 KB)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 11  bytes 2096 (2.0 KB)
        TX errors 0  dropped 2 overruns 0  carrier 0  collisions 0

I also did a tcpdump and found no udp message source from "fdc5:61d5:bd6e:1:c441:9814:bcaa:e476" and i tried others as well.

Please advise.

Regards,
dk

chshu, 2025-03-06:
@dk77148 everything looks good then, the UDP message is recevied by rcp, and forwarded to otbr-agent, did you see the increase of RX packets on wpan0 interface:

```
RX packets 1552 bytes 139863 (139.8 KB)
```

If yes, it maybe some issue with the tcpdump? Noticed that you are using the `VMware-Virtual-Platform`, did you run the tcpdump in the same environment? Could you check if it could capture other IP frames?

donaldkohcw, 2025-03-06:
Hi @chshu,

Yes, it increased over time. I did a "sudo tcpdump -i wpan0 -w capture4.pcap" but there is nothing from address "fd3a:231d:458b:9544:97b6:23d9:8862:5103" but only appear to have from "fe80::490:a449:353d:3743" as shown in the attachment. It shows that tcpdump has no issue. Do you have any idea? 

[image]

Regards,
dk

chshu, 2025-03-06:
> but only appear to have from "fe80::490:a449:353d:3743" as shown in the attachment.

It captured the frames sent from the Linux. Could you try to ping from the ot-cli to otbr, and check if the tcpdump can capture the ICMP6 frame?

donaldkohcw, 2025-03-06:
Hi @chshu,

Please find the outputs of ifconfig, ip6tables -vL, and cat /etc/default/otbr-agent below. 

I also disabled the firewall and reran the test, but the issue persists. The otbr-agent is the latest checkout from the Git repository. I did try it on another virtual machine and encountered the same problem, but I have yet to test it on a Linux PC. 

IFCONFIG:

donaldkoh@donaldkoh-VMware-Virtual-Platform:~/ot-br-posix$ ifconfig
ens33: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.131.128  netmask 255.255.255.0  broadcast 192.168.131.255
        inet6 fe80::20c:29ff:fe8e:b16  prefixlen 64  scopeid 0x20<link>
        inet6 fd73:4df9:15b:3185:20c:29ff:fe8e:b16  prefixlen 64  scopeid 0x0<global>
        inet6 fd73:4df9:15b:3185:45c3:c8e5:7ae4:8dec  prefixlen 64  scopeid 0x0<global>
        ether 00:0c:29:8e:0b:16  txqueuelen 1000  (Ethernet)
        RX packets 78336  bytes 105232221 (105.2 MB)
        RX errors 0  dropped 21  overruns 0  frame 0
        TX packets 40572  bytes 3009262 (3.0 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 1251  bytes 136406 (136.4 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1251  bytes 136406 (136.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wpan0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1280
        inet6 fd3a:231d:458b:9544:0:ff:fe00:fc11  prefixlen 64  scopeid 0x0<global>
        inet6 fd3a:231d:458b:9544:0:ff:fe00:6c00  prefixlen 64  scopeid 0x0<global>
        inet6 fd3a:231d:458b:9544:0:ff:fe00:fc10  prefixlen 64  scopeid 0x0<global>
        inet6 fd3a:231d:458b:9544:0:ff:fe00:fc38  prefixlen 64  scopeid 0x0<global>
        inet6 fd3a:231d:458b:9544:97b6:23d9:8862:5103  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::4490:a449:353d:3743  prefixlen 64  scopeid 0x20<link>
        inet6 fdc5:61d5:bd6e:1:c441:9814:bcaa:e476  prefixlen 64  scopeid 0x0<global>
        inet6 fd3a:231d:458b:9544:0:ff:fe00:fc00  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 2717  bytes 243660 (243.6 KB)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 17  bytes 2432 (2.4 KB)
        TX errors 0  dropped 2 overruns 0  carrier 0  collisions 0

IP6TABLES -vL:

onaldkoh@donaldkoh-VMware-Virtual-Platform:~/ot-br-posix$ sudo ip6tables -vL
Chain INPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 OTBR_FORWARD_INGRESS  all  --  any    wpan0   anywhere             anywhere            

Chain OUTPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain OTBR_FORWARD_INGRESS (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 DROP       all  --  wpan0  any     anywhere             anywhere             PKTTYPE = unicast
    0     0 DROP       all  --  any    any     anywhere             anywhere             match-set otbr-ingress-deny-src src
    0     0 ACCEPT     all  --  any    any     anywhere             anywhere             match-set otbr-ingress-allow-dst dst
    0     0 DROP       all  --  any    any     anywhere             anywhere             PKTTYPE = unicast
    0     0 ACCEPT     all  --  any    any     anywhere             anywhere            

cat /etc/default/otbr-agent:

# Default settings for otbr-agent. This file is sourced by systemd

# Options to pass to otbr-agent
#OTBR_AGENT_OPTS="-I wpan0 -B ens33 spinel+hdlc+uart:///dev/ttyACM0 trel://ens33"
#OTBR_NO_AUTO_ATTACH=0

OTBR_AGENT_OPTS="-I wpan0 -B ens33 spinel+hdlc+uart:///dev/ttyUSB0?uart-baudrate=460800"

Regards,
dk