espressif/esp-thread-br #136: Compatibility with ESP32-P4 DevKit ?

FnxQT, 2025-03-28:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello,

Is this thread border router integration compatible with the ESP32-P4 DevKit ? It seems to have everything needed, the P4 as the main chip and the C6 for thread.

Thanks.

FnxQT, 2025-03-29:
Thank you for the information. However, if I understand correctly, there is no plan to support the ESP32-P4 DevKit (with the ethernet module and the C6) as a thread border router ? I mean it would be the ideal platform. Or can we expect to have a modular Ethernet based solution like : P4 + Ethernet Module + C6 (RCP)

If not possible, do you think another example hardware platform will be provided ?

chshu, 2025-03-31:
@FnxQT The [ESP32-P4-Function-EV-Board](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32p4/esp32-p4-function-ev-board/index.html) supports `P4 + Ethernet Module + C6 (RCP)`.

FnxQT, 2025-03-31:
@chshu So can you confirm to me that I just need to wait end of May for the openthread border router libs to release for the P4 and then I can use the [ESP32-P4-Function-EV-Board](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32p4/esp32-p4-function-ev-board/index.html) as a border router between Ethernet and Thread ?

Thank you very much, I just want to be really sure I understand everything completely.

chshu, 2025-03-31:
> So can you confirm to me that I just need to wait end of May for the openthread border router libs to release for the P4 and then I can use the [...]

Yes, it's the plan.

chegewara, 2025-04-16:
Hi,
sorry, but i am lame. I am trying to build basic example on P4 and i am getting this error
```
ninja: error: '/home/chegewara/programming/esp-idf/esp32c5/components/openthread/lib/esp32p4/libopenthread_br.a', needed by 'esp_ot_br.elf', missing and no known rule to make it
```

I am using esp-idf master branch updated just now.

Thanks

gytxxsy, 2025-04-18:
> Hi, sorry, but i am lame. I am trying to build basic example on P4 and i am getting this error
>
> ```
> ninja: error: [...]

Hello, we currently do not have ESP OpenThread code that supports P4. As discussed in this issue, the work is in progress.

chegewara, 2025-04-23:
> [@FnxQT](https://github.com/FnxQT) The [ESP32-P4-Function-EV-Board](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32p4/esp32-p4-function-ev-board/index.html) supports `P4 + Ethernet Module + C6 (RCP)`.

i thought this is suggesting its already implemented, sorry

tanyanquan, 2025-05-20:
Hi @FnxQT and @chegewara , thank you very much for your patience. We are happy to let you know that we the [P4 thread-lib](https://github.com/espressif/esp-thread-lib/tree/459448381b89b7c8e82480e4836d0dc32eb9c68b) is now ready, and you can try out using P4 as a Thread Border Router using the [example in esp-idf](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br) from v5.3 onwards. 

If you are using the [ESP32-P4-Function-EV-Board](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32p4/esp32-p4-function-ev-board/index.html), you can either use the built-in ESP32-C6-MINI-1 as RCP, or a standalone RCP device (as described in the README). 

### Option 1: built-in ESP32-C6-MINI-1 as RCP (Ethernet only)

1. Flash the ESP32-C6 with the [ot_rcp firmware](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp) using [ESP_PROG](https://docs.espressif.com/projects/esp-iot-solution/en/latest/hw-reference/ESP-Prog_guide.html), instructions can be found [here](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/esp32_p4_function_ev_board.md#52-using-esp-prog)
2. Connect the TX and RX pins for the ESP32-C6 to the respective GPIO pins of the ESP32-P4 (default = 4 and 5)
3. Note that when the ESP32-C6 is being used as Thread RCP, only Ethernet connection is supported for Thread BR because ESP32-P4 does not have native Wi-Fi capability (therefore you will also see that `CONFIG_OPENTHREAD_BR_AUTO_START` is turned on in `sdkconfig.defaults`)
4. Note also that `esp-thread-lib` for ESP32-P4 is compiled with LwIP's PPP support turned on to support ESP-Hosted Wi-Fi (described below), so it is also turned on by default in `sdkconfig.defaults`

### Option 2: standalone RCP device (Wi-Fi or Ethernet)

1. Follow the instructions [here](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/esp32_p4_function_ev_board.md) for adding ESP-Hosted support to the ot_br example, by using the built-in ESP32-C6-MINI-1 as Wi-Fi co-processor (the communication between ESP32-P4 and ESP32-C6 will be done using build-in SDIO connections, so no additional connections are required)
2. Use a standalone H2 or C6 flashed with ot_rcp example as Thread RCP (connected via UART to ESP32-P4)

We are currently preparing the [basic_thread_border_router example](https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router) found in this repo to support RCP auto-update for ESP32-C6 to simplify Option 1 Step 1 for the ESP32-P4-Function-EV-Board (i.e. no need to Flash the ESP32-C6 using ESP-PROG). 

Do let us know if you face any issues and we will be happy to assist!

zwx1995esp, 2025-06-05:
Hi @chegewara and @FnxQT Please have a try with these two options. I will close this issue but feel free to reopen if you encounter any issues.

closed 2025-06-05 by zwx1995esp