espressif/esp-thread-br #138: Support RMII Ethernet Device

orangeagain, 2025-05-20:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Support more Ethernet Device example ip101, jusk like ESP32-Ethernet-Kit support RMII
https://github.com/espressif/esp-idf/tree/af25eb44/examples/ethernet/basic

### Use cases

SPI Ethernet is too expensive
For support more general-purpose interfaces to promote the Espressif Thread series more broadly, similar to what the ESP32-Ethernet-Kit has done.

orangeagain, 2025-05-20:
Saddly,same like esp32-c6 not support RMII:

Note

Although ESP32-C6 has no integrated Ethernet MAC, it is still possible to calculate an Ethernet MAC address. However, this MAC address can only be used with an external ethernet interface such as an SPI-Ethernet device. See [Ethernet]

https://docs.espressif.com/projects/esp-idf/en/stable/esp32c6/api-reference/system/misc_system_api.html

gytxxsy, 2025-05-22:
@orangeagain 
In fact, the `ESP32-S3` does not support Ethernet MAC and RMII. You can use the [ESP32-P4-Function-EV-Board](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32p4/esp32-p4-function-ev-board/index.html) which does support Ethernet MAC and RMII. You can refer to what is mentioned in this issue: https://github.com/espressif/esp-thread-br/issues/136#issuecomment-2893233649

zwx1995esp, 2025-06-19:
No, esp32s3 does not support EMAC, only ESP32-P4 and ESP32 supported. 
Please see:
[esp32 SOC capabilities line 78](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/include/soc/soc_caps.h#L78) and [esp32p4 SOC capabilities line 43](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32p4/include/soc/soc_caps.h#L43), but no this macro in [esp32s3 SOC capabilities](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32s3/include/soc/soc_caps.h)

zwx1995esp, 2025-06-19:
Also, please refer to our TRM, for [esp32 TRM](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf), chipter 24 Connectivity Interface. But no EMAC for [esp32s3 TRM](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf).