espressif/esp-thread-br #139: Request: Web GUI Video Walkthrough and Mobile App Availability for esp-thread-br

kunjsn123, 2025-06-14:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Here's your **revised GitHub issue** including the **mobile app availability query**:

---

### 📋 **Title:**

Request: Web GUI Video Walkthrough and Mobile App Availability for `esp-thread-br`

---

### 📝 **Body:**

Hello Espressif Team,

I'm actively working with the [esp-thread-br](https://github.com/espressif/esp-thread-br) project and using the Web GUI as documented here:
🔗 [[esp-thread-br Web GUI Docs](https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/web-gui.html)](https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/web-gui.html)

After exploring it in depth, I found that:

---

#### 🔍 Missing Information / Features:

* No clear **video walkthrough or demo** of how to use the Web GUI
* Limited documentation on:

  * **Deleting** or managing an existing Thread network
  * **Advanced network settings** through the GUI
  * Full interpretation of **topology, logs, and statuses**

---

#### 📷 Visual Comparison:

I’ve attached a screenshot (possibly from Nordic Semiconductor’s tools) that shows more comprehensive topology and control options. The current GUI from Espressif, while functional, lacks some of those features or documentation to use them effectively.

---

#### 📱 Question: Is there any Mobile App available?

Is there a mobile application (Android/iOS) that can be used to:

* View or control the Thread network?
* Commission devices via Bluetooth or QR codes?
* Monitor network topology or device states?

If such an app is not available now, are there any **plans** or **alternatives** (e.g., third-party tools) that you recommend?

---

### 🙏 Suggestion:

Please consider:

* Creating a **video demo** or screen-recorded walkthrough for the Web GUI.
* Expanding the documentation to include:

  * Deletion/reset operations
  * Network status interpretation
  * App/mobile integration guidance (if applicable)

Thank you for your amazing work in enabling Thread on ESP platforms! This kind of support would greatly help developers and integrators working in the ecosystem.

[image]

kunjsn123, 2025-06-18:
> Thank you for your suggestion. We will consider adding more detailed documentation.

Thank you for the update!

Could you please share a tentative timeline for when the enhanced documentation and features—especially the Web GUI improvements and potential mobile app support—will be available?

I'm currently developing a Thread-based system using ESP as part of a larger deployment project at a leading home automation company. These features (especially mobile commissioning and advanced GUI control) are critical for our integration.

If there’s no expected support in the near term, we may need to re-evaluate our platform choice and potentially consider Nordic Semiconductor, which already offers a more complete toolchain.

Your prompt response will help us make an informed decision. Looking forward to your clarification.

Thanks again for your great work so far!

kunjsn123, 2025-06-18:
Thank you for your response.

To clarify, my primary requirement is around **Thread device commissioning**. I understand this falls under the application layer and is not handled directly by the TBR SDK. However, the documentation link you provided:

🔗 [https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/web-gui.html#web-gui-application-introduction](https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/web-gui.html#web-gui-application-introduction)

...shows several UI elements implying features such as:

* Display of joined Thread devices
* Indication of whether a device is commissioned or not
* Network/device management controls (e.g., remove device from network)

These features are referenced visually, but **the actual APIs or methods to perform these actions are not documented**. My current blockers are:

---

### 🔧 Specific Technical Questions:

1. **Commissioning Devices**:
   Is there a REST API endpoint available (like in `ot-ctl` or `otbr-agent`) for commissioning Thread devices (e.g., via PSKd or QR)? If yes, please provide an example `curl` or code snippet for commissioning a device using PSKd.

2. **Listing Devices**:
   How can we retrieve the list of currently joined Thread devices via REST API? I’m looking for a way to replicate the information shown in the Web GUI under "Joined Devices".

3. **Removing Devices**:
   Is there an endpoint or CLI method exposed through the TBR or Web GUI backend to remove a specific child/leader/router device from the Thread network?

4. **Rainmaker + REST API**:
   You mentioned that customers typically build their own GUI. In the context of Rainmaker, how can one securely bridge and expose these REST APIs (e.g., via Node.js/Flask proxy) for cloud-based commissioning or management?

---

If these operations are not yet fully exposed through REST, is there any roadmap or internal method (even if experimental) we can use for development purposes?

Thanks again for your help. This is critical for evaluating continued development on ESP vs Nordic-based toolchains.

kunjsn123, 2025-07-23:
> It seems you're trying to extend TBR’s scope with application-layer capabilities, specifically device commissioning and management, which go beyond its intended role. I'd recommand you to consider Matter or Rainmaker solutions, you may also implement your own Applicaiton.
> 
> > Commissioning Devices:
> > Is there a REST API endpoint available (like in ot-ctl or otbr-agent) for commissioning Thread devices (e.g., via PSKd or QR)? If yes, please provide an example curl or code snippet for commissioning a device using PSKd.
> 
> No, commissioning cannot be performed via REST API. This is handled by the Commissioner (typically a mobile app). In this context, the TBR acts only as the Border Agent. You can refer to the external commissioning guide here: https://openthread.io/guides/border-router/external-commissioning
> 
> > Listing Devices:
> > How can we retrieve the list of currently joined Thread devices via REST API? I’m looking for a way to replicate the information shown in the Web GUI under "Joined Devices".
> 
> You can use the `http://192.168.200.98:80/topology` to get the network topology, more details here: https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/web-gui.html#web-gui-rest-apis
> 
> > Removing Devices:
> > Is there an endpoint or CLI method exposed through the TBR or Web GUI backend to remove a specific child/leader/router device from the Thread network?
> 
> No, the Thread protocol does not support removing devices through the network layer. Any such behavior must be implemented at the application level.
> 
> > Rainmaker + REST API:
> > You mentioned that customers typically build their own GUI. In the context of Rainmaker, how can one securely bridge and expose these REST APIs (e.g., via Node.js/Flask proxy) for cloud-based commissioning or management?
> 
> Rainmaker does not use Thread commissioning or TBR's REST APIs. It follows its own network provisioning protocol.
> 
> For further assistance, feel free to reach out to our support team and share your detailed requirements: https://www.espressif.com/en/contact-us/technical-inquiries.

Thank you for your detailed response and clarifications.

Regarding the last point you mentioned:

> *"You may also implement your own Application."*

I understand and agree that commissioning in the Thread ecosystem—especially when using the ESP Thread Border Router (TBR)—is considered an application-layer responsibility. However, I’d like to clarify that the external commissioning guide you linked ([https://openthread.io/guides/border-router/external-commissioning](https://openthread.io/guides/border-router/external-commissioning)) appears to be tailored for Nordic Semiconductor devices and their tooling (such as nRF52840 dongles, PC tools, etc.).

Since I'm working exclusively with **Espressif-based Thread devices** (ESP32-H2, ESP32-S3 for Border Router), I’m specifically looking for **Espressif-only guidance or examples** for implementing application-layer commissioning (via PSKd or QR, using ESP devices only, without Nordic tools or dependencies).

### My Follow-up Questions:

1. **Is there any Espressif-specific example, documentation, or guidance** for implementing an external commissioner app (e.g., mobile or desktop) that can interact with the ESP Thread Border Router for commissioning ESP Thread Joiner devices?

2. If no such reference exists yet, could you suggest the best way to **interface with the Border Agent** (e.g., via CoAP over UDP) to initiate commissioning manually from a custom application layer?

3. Since there is no REST API support for commissioning:

   * Is there any plan to support this in the future from Espressif’s Web GUI or ESP SDK side?
   * Or is building our own commissioner (possibly wrapping `libcommissioner` or `ot-commissioner`) the only current viable path?

We are evaluating Thread-based deployment at scale using Espressif's ecosystem and want to build the correct foundation now, especially around device onboarding and provisioning.

Thank you once again for the support—if there's any additional reference or internal example for the above, it would be incredibly helpful.

chshu, 2025-07-24:
>  I’d like to clarify that the external commissioning guide you linked (https://openthread.io/guides/border-router/external-commissioning) appears to be tailored for Nordic Semiconductor devices and their tooling (such as nRF52840 dongles, PC tools, etc.).

The commissioning guide applies to all Thread platforms running the OpenThread stack, it is not specific to any particular platform.

The OpenThread community maintains the external Commissioner app here:
https://github.com/openthread/ot-commissioner/tree/main/android

REST API support for commissioning will not be provided, as it does not comply with the Thread commissioning specification. The Commissioner and Border Agent are required to communicate over a DTLS session established using the PSKc.

closed 2025-11-05 by kunjsn123