espressif/esp-thread-br #147: [Bug] Ethernet crash on ESP32-S3/H2 combo module in esp-thread-br basic_thread_border_router example

fir3drag0n, 2025-08-03:
# Crash when using Ethernet mode on ESP32-S3/H2 combo module with esp-thread-br

**Description:**

I am using the `basic_thread_border_router` example from the latest `esp-thread-br` repository with an ESP32-S3 + ESP32-H2 combo module as RCP.

- ESP-IDF version: v5.5
- esp-thread-br branch: main
- Hardware: ESP32-S3 (host) + ESP32-H2 (RCP via UART)
- Example: `examples/basic_thread_border_router`
- Mode: Ethernet (W5500)

When I configure the project to use Ethernet (as described in the documentation) and flash it, the device crashes as soon as the Ethernet link comes up and an IP address is assigned.

**Log excerpt:**

```
I (3438) obtr_web: http://192.168.1.111:80/index.html
I (3438) esp_netif_handlers: example_netif_eth ip: 192.168.1.111, mask: 255.255.252.0, gw: 192.168.1.1
I (3438) ethernet_connect: Got IPv4 event: Interface "example_netif_eth" address: 192.168.1.111

assert failed: xQueueGenericSend queue.c:936 (pxQueue)

Backtrace: 0x40375b1d:0x3fcee6f0 0x4037c0b9:0x3fcee710 0x40383481:0x3fcee730 0x4037c41f:0x3fcee850 0x4201b1e7:0x3fcee890 0x420cbf83:0x3fcee8c0 0x420cbff3:0x3fcee910
...
--- 0x4201b1e7: eth_on_got_ip at C:/Users/.../protocol_examples_common/eth_connect.c:43
--- (inlined by) eth_on_got_ip at C:/Users/.../protocol_examples_common/eth_connect.c:35
```

The crash repeats after every reboot. Using Wi-Fi mode works fine on the same hardware.

**Steps to reproduce:**
1. Clone `esp-thread-br` and configure `basic_thread_border_router`
2. Set to use Ethernet (W5500)
3. Build and flash on ESP32-S3/H2 combo module
4. Connect W5500 to network and power up
5. Observe crash when IP is assigned

**Expected behavior:**
The Border Router should initialize and run in Ethernet mode without crashing.

**Actual behavior:**
Device reboots with assertion failure in `xQueueGenericSend` during `eth_on_got_ip`.

zwx1995esp, 2025-08-04:
Hi, @fir3drag0n  thanks for reporting this issue. Could you please share more informations?

1. Could you please share the commit hash of the IDF branch and esp-thread-br branch you used?
2. Could you please share the total logs? Expecially the whole backtrace logs.
3. Could you please share the sdkconfig you used?

fir3drag0n, 2025-08-04:
I think it is related to https://docs.espressif.com/projects/esp-idf/en/stable/esp32h2/api-reference/network/esp_eth.html#connect-driver-to-stack

zwx1995esp, 2025-10-21:
Hi @fir3drag0n, may I ask if this issue has been resolved on your side? Since there’s been no response for a while, I’ll go ahead and close this issue for now.

Please feel free to reopen it if the problem still persists