espressif/esp-thread-br #150: Unexpected RCP reset (RESET_OTHER) and repeated "dropping radio frame: Parse" errors when using ESP32-S3 + ESP32-C6 Border Router

kunjsn123, 2025-08-12:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expected the RCP (ESP32-C6) to operate continuously without unexpected resets, and UDP multicast messages to ff02::1 to be sent successfully without triggering errors or aborted transmissions.

### Actual behavior (suspected bug)

Instead, the RCP resets unexpectedly with RESET_OTHER while showing repeated

_**dropping radio frame: Parse**_

messages. After each reset, the Border Router attempts recovery (RCP recovery is done) but the error reappears shortly after. This results in dropped UDP multicast packets and interrupted communication.

### Error logs or terminal output

```shell
E (470699) OPENTHREAD: dropping radio frame: Parse
E(470700) OPENTHREAD:[C] P-RadioSpinel-: Unexpected RCP reset: RESET_OTHER
W(470701) OPENTHREAD:[W] P-RadioSpinel-: RCP failure detected
W(470702) OPENTHREAD:[W] P-RadioSpinel-: Trying to recover (1/3)
E(470705) OPENTHREAD:[C] P-SpinelDrive-: Software reset co-processor successfully
I(470723) OPENTHREAD:[N] P-RadioSpinel-: RCP recovery is done
I(470766) OPENTHREAD:[N] MeshForwarder-: Failed to send IPv6 UDP msg, len:90, chksum:62f7, ecn:no, to:0xffff, sec:no, error:Abort, prio:net
I(470768) OPENTHREAD:[N] MeshForwarder-:     src:[fe80:0:0:0:c8a9:bd44:d392:5380]:19788
I(470769) OPENTHREAD:[N] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788
```

### Steps to reproduce the behavior

Set up ESP32-S3 as Thread Border Router host (esp-thread-br example) and ESP32-C6 as RCP (RCP firmware from esp-thread-br).

Start the Border Router application.

Send periodic UDP multicast packets to ff02::1 port 19788 from Wi-Fi or Thread side.

Within a few minutes, observe RCP reset events and repeated error logs.

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu 24

### Shell

other (details in Additional context)

### Additional context

The problem seems correlated with frequent UDP multicast to ff02::1, but resets may also occur with other traffic.

This may be related to buffer handling or RCP transport stability when processing multicast frames.

The issue occurs even with light traffic load, suggesting it’s not just congestion.

zwx1995esp, 2025-10-21:
Hi @kunjsn123 may I ask that do you have any updates on this issue?

kunjsn123, 2025-10-25:
> [@kunjsn123](https://github.com/kunjsn123) Are you able to reproduce the issue on the [ESP Thread BR board](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html#wi-fi-based-thread-border-router) using the latest code base? If yes, could you share us the IDF commit ID and reproduce steps?

 Hi @chshu, thanks for the response.

 I can still reproduce the issue using the latest `esp-thread-br` example from the ESP-IDF main branch (v5.4.x) and the default RCP firmware for ESP32-C6.

 **My setup:**

 * Host: ESP32-S3 dev board running the Thread Border Router
 * RCP: ESP32-C6 dev board running `ot-rcp`
 * Connection: only **5V, GND, RST, TX, RX** (no CTS/RTS lines connected)

 **Observed behavior:**
 When I send periodic UDP multicast packets to `ff02::1` (port 19788), I get repeated

 ```
 E (470699) OPENTHREAD: dropping radio frame: Parse  
 E (470700) OPENTHREAD:[C] P-RadioSpinel-: Unexpected RCP reset: RESET_OTHER  
 ```

 followed by

 ```
 RCP recovery is done
 ```

 The issue then reappears shortly after.

 **Questions:**

 1. Is hardware flow control (CTS/RTS) required or strongly recommended between ESP32-S3 and ESP32-C6 for stable UART RCP communication?
 2. If yes, could you confirm which pins should be connected on both sides?
 3. If CTS/RTS are optional, what UART baud rate is considered stable without flow control?

 I suspect the missing flow control might be causing UART buffer overflow, leading to `Parse` errors and `RESET_OTHER` events. Please confirm if that’s the expected behavior when only TX/RX are used.

 Thanks!

zwx1995esp, 2025-10-30:
@gytxxsy PTAL for this issue.

gytxxsy, 2025-11-03:
@kunjsn123 
Regarding your three questions:
1. In our standard example, UART usage only includes `TX/RX`, with `EN/BOOT` used for RCP update. `CTS/RTS` are not used in our example.
2. When using UART, six connections (`TX, RX, EN, BOOT, VCC, GND`) are sufficient for running the border router.
3. The UART baud rate should be set to `460800`.

I tested this issue based on IDF `v5.4.2` and IDF `v5.4.3`, but could not reproduce it. Could you please provide more detailed reproduction steps?
Alternatively, I have several debugging suggestions and questions:
1. From the logs you provided, it appears that the RCP experienced an unexpected reboot, which doesn’t seem likely to be caused by sending broadcast packets. Have you tried using another ESP32-C6 or an ESP32-H2 to reproduce the issue, to confirm that it’s not hardware-related?
2. The address `ff02::1` and port `19788` are quite special — OpenThread itself uses them to broadcast MLE messages. What exactly do you mean by “UDP multicast packets to ff02::1 (port 19788)”? Are you actively sending custom messages from BR to ff02::1 (port 19788) on the Thread or Wi-Fi devices?

closed 2025-11-05 by kunjsn123