espressif/esp-thread-br #151: P4 main processor + C5 Wi-Fi co-processor + C5 Thread radio co-processor

dmartauz, 2025-08-13:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

We are considering P4 main processor + C5 Wi-Fi co-processor + C5 Thread radio co-processor combination on custom board due to following benefits:
- reduced BOM (identical module type for both RCPs)
- ANT2 RF pad of ESP32-C5-WROOM-1U allows to use on-board chip antenna

Is this combination feasible?

dmartauz, 2025-08-20:
@chshu yes, you are right, however H2 module does not allow to route RF trace to the PCB

Could you please advise on selection External coexistence GPIO pins on C5 for WiFi and C5 for Thread? Are there any GPIOs preferred?

chshu, 2025-08-20:
> Could you please advise on selection External coexistence GPIO pins on C5 for WiFi and C5 for Thread? Are there any GPIOs preferred?

Any generic GPIO can be configured as an external coex GPIO, so there is no specific preference